ros2/kilted_tutorial_party #1: Access Controls

tfoote, 2025-05-01:
Check the documentation for the 'Access Controls' page

## Setup
- Rmw: Fastdds
- BuildType: Debian
- Os: Noble
- Chip: Amd64

## Links
- [Access Controls page](https://docs.ros.org/en/kilted/Tutorials/Advanced/Security/Access-Controls.html)

## Checks
- [ ] **I was able to follow the documentation.**
- [ ] **The documentation seemed clear to me.**
- [ ] **The documentation didn't have any obvious errors.**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

SuperGops7, 2025-05-04:
I would like to test this @tfoote @ahcorde

SuperGops7, 2025-05-05:
Hi, I'm running into the following error with FastDDS

[image]

I did follow the steps to build the security plugins specified in the [ROS2 Security Introduction](https://docs.ros.org/en/kilted/Tutorials/Advanced/Security/Introducing-ros2-security.html), as below

`colcon build --symlink-install --cmake-args -DSECURITY=ON --packages-select fastdds rmw_fastrtps_cpp rmw_fastrtps_dynamic_cpp rmw_fastrtps_shared_cpp`

However, I'm still running into the same issue

SuperGops7, 2025-05-07:
@ahcorde @tfoote Can you help me with this issue?

SuperGops7, 2025-05-09:
Got it @mjcarroll , so can I move onto looking into another issue?

cottsay, 2025-05-13:
The debs have been rebuilt, so you should be able to update and run this tutorial now.

Aniketkumarroy, 2025-05-13:
Hi @cottsay I have tested this
### ***Writing our custom `permissions.xml` file***
[image]

### ***Using sros2 sample policies template***
[image]

if everything is alright we can close this

cottsay, 2025-05-13:
Thank you both for testing!

closed 2025-05-13 by cottsay