ros2/kilted_tutorial_party #10: Building a Movable Robot Model with URDF

tfoote, 2025-05-01:
Check the documentation for the 'Building a Movable Robot Model with URDF' page

## Setup
- Rmw: Fastdds
- BuildType: Debian
- Os: Noble
- Chip: Amd64

## Links
- [Building a Movable Robot Model with URDF page](https://docs.ros.org/en/kilted/Tutorials/Intermediate/URDF/Building-a-Movable-Robot-Model-with-URDF.html)

## Checks
- [ ] **I was able to follow the documentation.**
- [ ] **The documentation seemed clear to me.**
- [ ] **The documentation didn't have any obvious errors.**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

4-legends, 2025-05-03:
Testing this ticket

4-legends, 2025-05-03:
- Setup
```
legends@4legends-ASUS:~/ros2_ws$ printenv | grep ROS
ROS_VERSION=2
ROS_PYTHON_VERSION=3
ROS_DOMAIN_ID=0
ROS_AUTOMATIC_DISCOVERY_RANGE=SUBNET
ROS_DISTRO=kilted

legends@4legends-ASUS:~/ros2_ws$ printenv | grep RMW
RMW_IMPLEMENTATION=rmw_fastrtps_cpp

``` 

- Issues
Related to RVIZ2 missing svg files that I have mentioned in #475 

- [x]  I was able to follow the documentation.
- [x] The documentation seemed clear to me.
- [x] The documentation didn't have any obvious errors.

https://github.com/user-attachments/assets/04fb037a-2436-4011-a5de-3f0289b78dc3

- Logs
```
legends@4legends-ASUS:~$ ros2 launch urdf_tutorial display.launch.py model:=urdf/06-flexible.urdf
[INFO] [launch]: All log files can be found below /home/legends/.ros/log/2025-05-03-11-31-01-495220-4legends-ASUS-14376
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [robot_state_publisher-1]: process started with pid [14380]
[INFO] [joint_state_publisher_gui-2]: process started with pid [14381]
[INFO] [rviz2-3]: process started with pid [14382]
[robot_state_publisher-1] [INFO] [1746297061.617335063] [robot_state_publisher]: Robot initialized
[rviz2-3] [INFO] [1746297061.803173861] [rviz2]: Stereo is NOT SUPPORTED
[rviz2-3] [INFO] [1746297061.803213960] [rviz2]: OpenGl version: 4.6 (GLSL 4.6)
[rviz2-3] [INFO] [1746297061.815635257] [rviz2]: Stereo is NOT SUPPORTED
[joint_state_publisher_gui-2] [INFO] [1746297061.963869409] [joint_state_publisher]: Waiting for robot_description to be published on the robot_description topic...
[joint_state_publisher_gui-2] [INFO] [1746297061.970082583] [joint_state_publisher]: Centering
[joint_state_publisher_gui-2] [INFO] [1746297062.131676883] [joint_state_publisher]: Centering
[rviz2-3] [ERROR] [1746297064.929361734] [rviz.ros_resource_retriever]: Timeout: Not able to call the service /rviz/get_resource
[rviz2-3] [ERROR] [1746297067.931433436] [rviz.ros_resource_retriever]: Timeout: Not able to call the service /rviz/get_resource
[rviz2-3] [ERROR] [1746297070.932729150] [rviz.ros_resource_retriever]: Timeout: Not able to call the service /rviz/get_resource
[rviz2-3] [ERROR] [1746297073.938954698] [rviz.ros_resource_retriever]: Timeout: Not able to call the service /rviz/get_resource
[rviz2-3] [ERROR] [1746297076.952908249] [rviz.ros_resource_retriever]: Timeout: Not able to call the service /rviz/get_resource
[rviz2-3] [ERROR] [1746297079.954026149] [rviz.ros_resource_retriever]: Timeout: Not able to call the service /rviz/get_resource
[rviz2-3] [ERROR] [1746297082.955707805] [rviz.ros_resource_retriever]: Timeout: Not able to call the service /rviz/get_resource
[rviz2-3] [ERROR] [1746297085.958781389] [rviz.ros_resource_retriever]: Timeout: Not able to call the service /rviz/get_resource
[joint_state_publisher_gui-2] [INFO] [1746297131.224215734] [joint_state_publisher]: Randomizing
[joint_state_publisher_gui-2] [INFO] [1746297132.897236052] [joint_state_publisher]: Centering
[joint_state_publisher_gui-2] [INFO] [1746297134.371571919] [joint_state_publisher]: Randomizing
[joint_state_publisher_gui-2] [INFO] [1746297135.693917114] [joint_state_publisher]: Randomizing
[joint_state_publisher_gui-2] [INFO] [1746297136.559706592] [joint_state_publisher]: Randomizing
[joint_state_publisher_gui-2] [INFO] [1746297137.345396492] [joint_state_publisher]: Randomizing
[joint_state_publisher_gui-2] [INFO] [1746297138.009416478] [joint_state_publisher]: Randomizing
[joint_state_publisher_gui-2] [INFO] [1746297138.761578329] [joint_state_publisher]: Randomizing
[joint_state_publisher_gui-2] [INFO] [1746297139.915507845] [joint_state_publisher]: Centering
^C[WARNING] [launch]: user interrupted with ctrl-c (SIGINT)
[rviz2-3] [INFO] [1746297296.478483072] [rclcpp]: signal_handler(signum=2)
[robot_state_publisher-1] [INFO] [1746297296.478489858] [rclcpp]: signal_handler(signum=2)
[ERROR] [joint_state_publisher_gui-2]: process has died [pid 14381, exit code -2, cmd '/opt/ros/kilted/lib/joint_state_publisher_gui/joint_state_publisher_gui --ros-args'].
[INFO] [rviz2-3]: process has finished cleanly [pid 14382]
[INFO] [robot_state_publisher-1]: process has finished cleanly [pid 14380]
```

closed 2025-05-05 by ahcorde