ros2/kilted_tutorial_party #100: Composition

tfoote, 2025-05-01:
Check the documentation for the 'Composition' page

## Setup
- Rmw: Fastdds
- BuildType: Debian
- Os: Noble
- Chip: Amd64

## Links
- [Composition page](https://docs.ros.org/en/kilted/Tutorials/Intermediate/Composition.html)

## Checks
- [ ] **I was able to follow the documentation.**
- [ ] **The documentation seemed clear to me.**
- [ ] **The documentation didn't have any obvious errors.**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

Amronos, 2025-05-03:
I followed https://docs.ros.org/en/kilted/Tutorials/Intermediate/Composition.html

- [x] I was able to follow the documentation.
- [x] The documentation seemed clear to me.
- [x] The documentation didn't have any obvious errors.

Some nitpicks:
1. In the [Unloading Components section](https://docs.ros.org/en/kilted/Tutorials/Intermediate/Composition.html#unloading-components), it isn't mentioned how one can go about getting the unique ID for each component.
2. In the [Passing parameter values into components](https://docs.ros.org/en/kilted/Tutorials/Intermediate/Composition.html#id16), a command to run ``rqt_image_view`` can be added.

Amronos, 2025-05-03:
> Some nitpicks:
>
>     1. In the [Unloading Components section](https://docs.ros.org/en/kilted/Tutorials/Intermediate/Composition.html#unloading-components), it isn't mentioned how one can go about [...]

Fixed with https://github.com/ros2/ros2_documentation/pull/5487.

One small thing about ``rqt_image_view``, when exiting it, I get the following warning:
```console
amronos@dev-pc:~$ ros2 run rqt_image_view rqt_image_view
Warning: class_loader.ClassLoader: SEVERE WARNING!!! Attempting to unload library while objects created by this loader exist in the heap! You should delete your objects before attempting to unload the library or destroying the ClassLoader. The library will NOT be unloaded.
         at line 127 in ./src/class_loader.cpp
```

ahcorde, 2025-05-12:
Thank you for testing

closed 2025-05-12 by ahcorde