ros2/kilted_tutorial_party #101: Composition executables

tfoote, 2025-05-01:
## Setup
- Rmw: Cyclone
- BuildType: Binary
- Os: Windows
- Chip: Amd64

## Links
- [Composition demos](https://github.com/ros2/demos/blob/kilted/composition/README.md)

## Checks
- [ ] **The executables in the README are present and work as expected.**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

ShahazadAbdulla, 2025-05-04:
I'll test this on Windows 11 / amd64 / binary / CycloneDDS

ShahazadAbdulla, 2025-05-04:
Testing Issue #101 on Windows 11 / amd64 / binary / CycloneDDS.

Here are the results for each check:

1.  **Manual Composition (`ros2 run composition manual_composition`):** SUCCESS. Output showed interleaved "Publishing:", "I heard:", "Incoming request:", and "Got result:" messages as expected.

2.  **DlOpen Composition (`ros2 run composition dlopen_composition ...`):** SUCCESS. Used the following command after getting the prefix:
    `ros2 run composition dlopen_composition C:\pixi_ws\ros2-windows\bin\talker_component.dll C:\pixi_ws\ros2-windows\bin\listener_component.dll`
    Output showed "Load library", "Instantiate class", "Publishing:", and "I heard:" messages as expected. *(Note: Mentioned path adjustment for clarity)*

3.  **Linktime Composition (`ros2 run composition linktime_composition`):** FAILED.
    *   The process started and printed the initial "[INFO] [... [linktime_composition]: Load library" message.
    *   However, no further output appeared (no "Instantiate class..." or communication messages).
    *   The process seemed stalled until interrupted with Ctrl+C.
    *   Output:
        ```
        (pixi_ros2_rolling) C:\pixi_ws>ros2 run composition linktime_composition
        [INFO] [1746352982.302522200] [linktime_composition]: Load library
        ```
       nothing happens after Load library just being stuck there

4.  **Composition Using Launch Actions (`ros2 launch composition composition_demo_launch.py`):** SUCCESS. Output showed launch messages, "Load Library", "Instantiate class", "Loaded node", "Publishing:", and "I heard:" messages as expected.

**Overall:** Most composition methods worked correctly, but the `linktime_composition` executable appears non-functional in this environment (Win11/amd64/binary/CycloneDDS).