ros2/kilted_tutorial_party #102: Composition executables

tfoote, 2025-05-01:
## Setup
- Rmw: Cyclone
- BuildType: Debian
- Os: Noble
- Chip: Amd64

## Links
- [Composition demos](https://github.com/ros2/demos/blob/kilted/composition/README.md)

## Checks
- [ ] **The executables in the README are present and work as expected.**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

jncfa, 2025-05-02:
Tested with Ubuntu Docker Container for amd64(under Rosetta 2) (ghcr.io/sloretz/ros-testing:kilted-desktop)

[image]

[image]

ahcorde, 2025-05-02:
Thank you for testing

closed 2025-05-02 by ahcorde