ros2/kilted_tutorial_party #103: Composition executables

tfoote, 2025-05-01:
## Setup
- Rmw: Fastdds
- BuildType: Binary
- Os: Windows
- Chip: Amd64

## Links
- [Composition demos](https://github.com/ros2/demos/blob/kilted/composition/README.md)

## Checks
- [ ] **The executables in the README are present and work as expected.**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

pradyum, 2025-05-03:
**Partially working**

---

### Manual Composition

```
C:\pixi_ws>ros2 run composition manual_composition
[INFO] [1746230865.518205700] [talker]: Publishing: 'Hello World: 1'
[INFO] [1746230865.523530500] [listener]: I heard: [Hello World: 1]
[INFO] [1746230866.518777200] [talker]: Publishing: 'Hello World: 2'
[INFO] [1746230866.521119000] [listener]: I heard: [Hello World: 2]
[INFO] [1746230866.590344100] [Server]: Incoming request: [a: 2, b: 3]
[INFO] [1746230866.590784900] [Client]: Got result: [5]
[INFO] [1746230867.518509000] [talker]: Publishing: 'Hello World: 3'
[INFO] [1746230867.522923700] [listener]: I heard: [Hello World: 3]
[INFO] [1746230868.518987600] [talker]: Publishing: 'Hello World: 4'
[INFO] [1746230868.521418400] [listener]: I heard: [Hello World: 4]
[INFO] [1746230868.566460000] [Server]: Incoming request: [a: 2, b: 3]
[INFO] [1746230868.573865200] [Client]: Got result: [5]
[INFO] [1746230869.518548000] [talker]: Publishing: 'Hello World: 5'
[INFO] [1746230869.521530300] [listener]: I heard: [Hello World: 5]
[INFO] [1746230870.519309600] [talker]: Publishing: 'Hello World: 6'
[INFO] [1746230870.523284900] [listener]: I heard: [Hello World: 6]
[INFO] [1746230870.565176600] [Server]: Incoming request: [a: 2, b: 3]
[INFO] [1746230870.569126600] [Client]: Got result: [5]
[INFO] [1746230871.518515900] [talker]: Publishing: 'Hello World: 7'
[INFO] [1746230871.520857300] [listener]: I heard: [Hello World: 7]
[INFO] [1746230872.518490100] [talker]: Publishing: 'Hello World: 8'
[INFO] [1746230872.521971700] [listener]: I heard: [Hello World: 8]
[INFO] [1746230872.565907100] [Server]: Incoming request: [a: 2, b: 3]
[INFO] [1746230872.569087500] [Client]: Got result: [5]
[INFO] [1746230873.518457200] [talker]: Publishing: 'Hello World: 9'
[INFO] [1746230873.521502600] [listener]: I heard: [Hello World: 9]
[INFO] [1746230874.518684500] [talker]: Publishing: 'Hello World: 10'
[INFO] [1746230874.519155900] [listener]: I heard: [Hello World: 10]
[INFO] [1746230874.566593300] [Server]: Incoming request: [a: 2, b: 3]
[INFO] [1746230874.570119000] [Client]: Got result: [5]
```

---
### DlOpen Composition

```
C:\pixi_ws\ros2-windows\bin>ros2 run composition dlopen_composition talker_component.dll listener_component.dll
[INFO] [1746231996.010033400] [dlopen_composition]: Load library talker_component.dll
[INFO] [1746231996.022431000] [dlopen_composition]: Instantiate class rclcpp_components::NodeFactoryTemplate<composition::Talker>
[INFO] [1746231996.248966200] [dlopen_composition]: Load library listener_component.dll
[INFO] [1746231996.259568600] [dlopen_composition]: Instantiate class rclcpp_components::NodeFactoryTemplate<composition::Listener>
[INFO] [1746231997.250597500] [talker]: Publishing: 'Hello World: 1'
[INFO] [1746231997.254478400] [listener]: I heard: [Hello World: 1]
[INFO] [1746231998.250495900] [talker]: Publishing: 'Hello World: 2'
[INFO] [1746231998.253520000] [listener]: I heard: [Hello World: 2]
[INFO] [1746231999.250873200] [talker]: Publishing: 'Hello World: 3'
[INFO] [1746231999.253679400] [listener]: I heard: [Hello World: 3]
[INFO] [1746232000.251178400] [talker]: Publishing: 'Hello World: 4'
[INFO] [1746232000.253504800] [listener]: I heard: [Hello World: 4]
[INFO] [1746232001.250278100] [talker]: Publishing: 'Hello World: 5'
[INFO] [1746232001.253331700] [listener]: I heard: [Hello World: 5]
[INFO] [1746232002.250971000] [talker]: Publishing: 'Hello World: 6'
[INFO] [1746232002.254653600] [listener]: I heard: [Hello World: 6]
```

---
### Linktime Composition

**I wasn't able to get it running in windows**

---
### Composition Using Launch Actions

```
C:\pixi_ws>ros2 launch composition composition_demo_launch.py
C:\pixi_ws\ros2-windows\Lib\site-packages\launch\logging\__init__.py:149: UserWarning: Cannot create a symlink to latest log directory: [WinError 1314] A required privilege is not held by the client: 'C:\\Users\\pc1\\.ros\\log\\2025-05-02-17-21-40-499180-DESKTOP-8BMD1U3-1132' -> 'C:\\Users\\pc1\\.ros\\log\\latest'

  warnings.warn(
[INFO] [launch]: All log files can be found below C:\Users\pc1\.ros\log\2025-05-02-17-21-40-499180-DESKTOP-8BMD1U3-1132
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [component_container.EXE-1]: process started with pid [4128]
[component_container.EXE-1] [INFO] [1746231701.662544800] [my_container]: Load Library: C:\pixi_ws\ros2-windows\bin/talker_component.dll
[component_container.EXE-1] [INFO] [1746231701.669216400] [my_container]: Found class: rclcpp_components::NodeFactoryTemplate<composition::Talker>
[component_container.EXE-1] [INFO] [1746231701.669295800] [my_container]: Instantiate class: rclcpp_components::NodeFactoryTemplate<composition::Talker>
[INFO] [launch_ros.actions.load_composable_nodes]: Loaded node '/talker' in container '/my_container'
[component_container.EXE-1] [INFO] [1746231701.691718100] [my_container]: Load Library: C:\pixi_ws\ros2-windows\bin/listener_component.dll
[component_container.EXE-1] [INFO] [1746231701.701188300] [my_container]: Found class: rclcpp_components::NodeFactoryTemplate<composition::Listener>
[component_container.EXE-1] [INFO] [1746231701.701259300] [my_container]: Instantiate class: rclcpp_components::NodeFactoryTemplate<composition::Listener>
[INFO] [launch_ros.actions.load_composable_nodes]: Loaded node '/listener' in container '/my_container'
[component_container.EXE-1] [INFO] [1746231702.693364000] [talker]: Publishing: 'Hello World: 1'
[component_container.EXE-1] [INFO] [1746231702.693740600] [listener]: I heard: [Hello World: 1]
[component_container.EXE-1] [INFO] [1746231703.693032300] [talker]: Publishing: 'Hello World: 2'
[component_container.EXE-1] [INFO] [1746231703.693347700] [listener]: I heard: [Hello World: 2]
[component_container.EXE-1] [INFO] [1746231704.692920500] [talker]: Publishing: 'Hello World: 3'
[component_container.EXE-1] [INFO] [1746231704.693153900] [listener]: I heard: [Hello World: 3]
```

ahcorde, 2025-05-05:
Can you also include the traces for `Linktime Composition` ?

Aniketkumarroy, 2025-05-13:
Hi @ahcorde I also tested this on Windows 11, here are the traces
### **Manual Composition**

[image]
### **DlOpen Composition**

[image]
### **Linktime Composition**

[image]
### **Composition Using Launch Actions**

[image]

ahcorde, 2025-05-13:
Thank you for testing

closed 2025-05-13 by ahcorde