ros2/kilted_tutorial_party #104: Composition executables

tfoote, 2025-05-01:
## Setup
- Rmw: Fastdds
- BuildType: Debian
- Os: Noble
- Chip: Amd64

## Links
- [Composition demos](https://github.com/ros2/demos/blob/kilted/composition/README.md)

## Checks
- [ ] **The executables in the README are present and work as expected.**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

Amronos, 2025-05-03:
- [x] The executables in the README are present and work as expected.
```console
amronos@dev-pc:~$ ros2 run composition manual_composition
[INFO] [1746276659.661291959] [talker]: Publishing: 'Hello World: 1'
[INFO] [1746276659.661868757] [listener]: I heard: [Hello World: 1]
[INFO] [1746276660.661149448] [talker]: Publishing: 'Hello World: 2'
[INFO] [1746276660.661352908] [listener]: I heard: [Hello World: 2]
[INFO] [1746276660.737643606] [Server]: Incoming request: [a: 2, b: 3]
[INFO] [1746276660.737873946] [Client]: Got result: [5]
[INFO] [1746276661.661174814] [talker]: Publishing: 'Hello World: 3'
[INFO] [1746276661.661376661] [listener]: I heard: [Hello World: 3]
[INFO] [1746276662.661104597] [talker]: Publishing: 'Hello World: 4'
[INFO] [1746276662.661285285] [listener]: I heard: [Hello World: 4]
[INFO] [1746276662.737500785] [Server]: Incoming request: [a: 2, b: 3]
[INFO] [1746276662.737634985] [Client]: Got result: [5]

amronos@dev-pc:~$ ros2 run composition dlopen_composition `ros2 pkg prefix composition`/lib/libtalker_component.so `ros2 pkg prefix composition`/lib/liblistener_component.so
[INFO] [1746276734.989309758] [dlopen_composition]: Load library /opt/ros/kilted/lib/libtalker_component.so
[INFO] [1746276734.990116655] [dlopen_composition]: Instantiate class rclcpp_components::NodeFactoryTemplate<composition::Talker>
[INFO] [1746276735.008855509] [dlopen_composition]: Load library /opt/ros/kilted/lib/liblistener_component.so
[INFO] [1746276735.009492810] [dlopen_composition]: Instantiate class rclcpp_components::NodeFactoryTemplate<composition::Listener>
[INFO] [1746276736.008970404] [talker]: Publishing: 'Hello World: 1'
[INFO] [1746276736.009309879] [listener]: I heard: [Hello World: 1]
[INFO] [1746276737.008948558] [talker]: Publishing: 'Hello World: 2'
[INFO] [1746276737.009198235] [listener]: I heard: [Hello World: 2]
[INFO] [1746276738.008944638] [talker]: Publishing: 'Hello World: 3'
[INFO] [1746276738.009212288] [listener]: I heard: [Hello World: 3]
[INFO] [1746276739.008974774] [talker]: Publishing: 'Hello World: 4'
[INFO] [1746276739.009171652] [listener]: I heard: [Hello World: 4]
[INFO] [1746276740.008932337] [talker]: Publishing: 'Hello World: 5'
[INFO] [1746276740.009133413] [listener]: I heard: [Hello World: 5]

amronos@dev-pc:~$ ros2 run composition linktime_composition
[INFO] [1746277134.474164813] [linktime_composition]: Load library 
[INFO] [1746277134.474215578] [linktime_composition]: Instantiate class rclcpp_components::NodeFactoryTemplate<composition::Client>
[INFO] [1746277134.830723374] [linktime_composition]: Instantiate class rclcpp_components::NodeFactoryTemplate<composition::Listener>
[INFO] [1746277134.836859311] [linktime_composition]: Instantiate class rclcpp_components::NodeFactoryTemplate<composition::Server>
[INFO] [1746277134.842906452] [linktime_composition]: Instantiate class rclcpp_components::NodeFactoryTemplate<composition::Talker>
[INFO] [1746277135.849132599] [talker]: Publishing: 'Hello World: 1'
[INFO] [1746277135.849469810] [listener]: I heard: [Hello World: 1]
[INFO] [1746277136.831293980] [Server]: Incoming request: [a: 2, b: 3]
[INFO] [1746277136.831560669] [Client]: Got result: [5]
[INFO] [1746277136.849158414] [talker]: Publishing: 'Hello World: 2'
[INFO] [1746277136.849483192] [listener]: I heard: [Hello World: 2]
[INFO] [1746277137.849101520] [talker]: Publishing: 'Hello World: 3'
[INFO] [1746277137.849308067] [listener]: I heard: [Hello World: 3]
[INFO] [1746277138.831013256] [Server]: Incoming request: [a: 2, b: 3]
[INFO] [1746277138.831145995] [Client]: Got result: [5]
[INFO] [1746277138.849102461] [talker]: Publishing: 'Hello World: 4'
[INFO] [1746277138.849306283] [listener]: I heard: [Hello World: 4]
[INFO] [1746277139.849046842] [talker]: Publishing: 'Hello World: 5'
[INFO] [1746277139.849249862] [listener]: I heard: [Hello World: 5]
[INFO] [1746277140.831563222] [Server]: Incoming request: [a: 2, b: 3]
[INFO] [1746277140.831766462] [Client]: Got result: [5]

amronos@dev-pc:~$ ros2 launch composition composition_demo_launch.py
[INFO] [launch]: All log files can be found below /home/amronos/.ros/log/2025-05-03-18-29-31-619510-dev-pc-33676
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [component_container-1]: process started with pid [33696]
[component_container-1] [INFO] [1746277172.108728963] [my_container]: Load Library: /opt/ros/kilted/lib/libtalker_component.so
[component_container-1] [INFO] [1746277172.110294344] [my_container]: Found class: rclcpp_components::NodeFactoryTemplate<composition::Talker>
[component_container-1] [INFO] [1746277172.110349697] [my_container]: Instantiate class: rclcpp_components::NodeFactoryTemplate<composition::Talker>
[INFO] [launch_ros.actions.load_composable_nodes]: Loaded node '/talker' in container '/my_container'
[component_container-1] [INFO] [1746277172.123115299] [my_container]: Load Library: /opt/ros/kilted/lib/liblistener_component.so
[component_container-1] [INFO] [1746277172.123797937] [my_container]: Found class: rclcpp_components::NodeFactoryTemplate<composition::Listener>
[component_container-1] [INFO] [1746277172.123819758] [my_container]: Instantiate class: rclcpp_components::NodeFactoryTemplate<composition::Listener>
[INFO] [launch_ros.actions.load_composable_nodes]: Loaded node '/listener' in container '/my_container'
[component_container-1] [INFO] [1746277173.120811149] [talker]: Publishing: 'Hello World: 1'
[... cut ...]
[component_container-1] [INFO] [1746277175.120826182] [talker]: Publishing: 'Hello World: 3'
[component_container-1] [INFO] [1746277175.121090567] [listener]: I heard: [Hello World: 3]
[component_container-1] [INFO] [1746277176.120818178] [talker]: Publishing: 'Hello World: 4'
[component_container-1] [INFO] [1746277176.121072174] [listener]: I heard: [Hello World: 4]
[component_container-1] [INFO] [1746277177.120751410] [talker]: Publishing: 'Hello World: 5'
[component_container-1] [INFO] [1746277177.120937559] [listener]: I heard: [Hello World: 5]
```

ahcorde, 2025-05-05:
Thank you for testing

closed 2025-05-05 by ahcorde