ros2/kilted_tutorial_party #107: Configure service introspection

tfoote, 2025-05-01:
Check the documentation for the 'Configure service introspection' page

## Setup
- Rmw: Fastdds
- BuildType: Debian
- Os: Noble
- Chip: Amd64

## Links
- [Configure service introspection page](https://docs.ros.org/en/kilted/Tutorials/Demos/Service-Introspection.html)

## Checks
- [ ] **I was able to follow the documentation.**
- [ ] **The documentation seemed clear to me.**
- [ ] **The documentation didn't have any obvious errors.**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

danielcranston, 2025-05-03:
- [x] **I was able to follow the documentation.**
- [x] **The documentation seemed clear to me.**
- [x] **The documentation didn't have any obvious errors.**

Looks good, I'll make a PR with some optional change suggestions.

christophebedard, 2025-05-03:
Thanks for testing

closed 2025-05-03 by christophebedard