ros2/kilted_tutorial_party #11: Building a Visual Robot Model with URDF from Scratch

tfoote, 2025-05-01:
Check the documentation for the 'Building a Visual Robot Model with URDF from Scratch' page

## Setup
- Rmw: Fastdds
- BuildType: Debian
- Os: Noble
- Chip: Amd64

## Links
- [Building a Visual Robot Model with URDF from Scratch page](https://docs.ros.org/en/kilted/Tutorials/Intermediate/URDF/Building-a-Visual-Robot-Model-with-URDF-from-Scratch.html)

## Checks
- [ ] **I was able to follow the documentation.**
- [ ] **The documentation seemed clear to me.**
- [ ] **The documentation didn't have any obvious errors.**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

4-legends, 2025-05-03:
Testing this ticket

4-legends, 2025-05-03:
- Setup
```
legends@4legends-ASUS:~/ros2_ws$ printenv | grep ROS
ROS_VERSION=2
ROS_PYTHON_VERSION=3
ROS_DOMAIN_ID=0
ROS_AUTOMATIC_DISCOVERY_RANGE=SUBNET
ROS_DISTRO=kilted

legends@4legends-ASUS:~/ros2_ws$ printenv | grep RMW
RMW_IMPLEMENTATION=rmw_fastrtps_cpp

``` 

- [ ]  I was able to follow the documentation.
- [ ] The documentation seemed clear to me.
- [ ] The documentation didn't have any obvious errors.

[image]

Logs:
```
legends@4legends-ASUS:~$ ros2 launch urdf_tutorial display.launch.py model:=urdf/01-myfirst.urdf
[INFO] [launch]: All log files can be found below /home/legends/.ros/log/2025-05-03-11-24-54-486078-4legends-ASUS-13723
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [robot_state_publisher-1]: process started with pid [13728]
[INFO] [joint_state_publisher_gui-2]: process started with pid [13729]
[INFO] [rviz2-3]: process started with pid [13730]
[robot_state_publisher-1] [INFO] [1746296694.604767340] [robot_state_publisher]: Robot initialized
[rviz2-3] [INFO] [1746296694.800801626] [rviz2]: Stereo is NOT SUPPORTED
[rviz2-3] [INFO] [1746296694.800839795] [rviz2]: OpenGl version: 4.6 (GLSL 4.6)
[rviz2-3] [INFO] [1746296694.813232233] [rviz2]: Stereo is NOT SUPPORTED
[joint_state_publisher_gui-2] [INFO] [1746296694.943261710] [joint_state_publisher]: Waiting for robot_description to be published on the robot_description topic...
[joint_state_publisher_gui-2] [INFO] [1746296694.949164991] [joint_state_publisher]: Centering
[joint_state_publisher_gui-2] [INFO] [1746296695.096116500] [joint_state_publisher]: Centering
^C[WARNING] [launch]: user interrupted with ctrl-c (SIGINT)
[rviz2-3] [INFO] [1746296697.736112921] [rclcpp]: signal_handler(signum=2)
[robot_state_publisher-1] [INFO] [1746296697.736263275] [rclcpp]: signal_handler(signum=2)
[ERROR] [joint_state_publisher_gui-2]: process has died [pid 13729, exit code -2, cmd '/opt/ros/kilted/lib/joint_state_publisher_gui/joint_state_publisher_gui --ros-args'].
[INFO] [rviz2-3]: process has finished cleanly [pid 13730]
[INFO] [robot_state_publisher-1]: process has finished cleanly [pid 13728]

legends@4legends-ASUS:~$ ros2 launch urdf_tutorial display.launch.py model:=urdf/02-multipleshapes.urdf
[INFO] [launch]: All log files can be found below /home/legends/.ros/log/2025-05-03-11-25-01-094217-4legends-ASUS-13845
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [robot_state_publisher-1]: process started with pid [13853]
[INFO] [joint_state_publisher_gui-2]: process started with pid [13854]
[INFO] [rviz2-3]: process started with pid [13855]
[robot_state_publisher-1] [INFO] [1746296701.210917965] [robot_state_publisher]: Robot initialized
[rviz2-3] [INFO] [1746296701.389437898] [rviz2]: Stereo is NOT SUPPORTED
[rviz2-3] [INFO] [1746296701.389477334] [rviz2]: OpenGl version: 4.6 (GLSL 4.6)
[rviz2-3] [INFO] [1746296701.403758786] [rviz2]: Stereo is NOT SUPPORTED
[joint_state_publisher_gui-2] [INFO] [1746296701.563904360] [joint_state_publisher]: Waiting for robot_description to be published on the robot_description topic...
[joint_state_publisher_gui-2] [INFO] [1746296701.570229348] [joint_state_publisher]: Centering
[joint_state_publisher_gui-2] [INFO] [1746296701.718033940] [joint_state_publisher]: Centering
^C[WARNING] [launch]: user interrupted with ctrl-c (SIGINT)
[rviz2-3] [INFO] [1746296703.326797585] [rclcpp]: signal_handler(signum=2)
[robot_state_publisher-1] [INFO] [1746296703.326855914] [rclcpp]: signal_handler(signum=2)
[ERROR] [joint_state_publisher_gui-2]: process has died [pid 13854, exit code -2, cmd '/opt/ros/kilted/lib/joint_state_publisher_gui/joint_state_publisher_gui --ros-args'].
[INFO] [robot_state_publisher-1]: process has finished cleanly [pid 13853]
[INFO] [rviz2-3]: process has finished cleanly [pid 13855]

legends@4legends-ASUS:~$ ros2 launch urdf_tutorial display.launch.py model:=urdf/03-origins.urdf
[INFO] [launch]: All log files can be found below /home/legends/.ros/log/2025-05-03-11-25-09-503057-4legends-ASUS-13936
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [robot_state_publisher-1]: process started with pid [13940]
[INFO] [joint_state_publisher_gui-2]: process started with pid [13941]
[INFO] [rviz2-3]: process started with pid [13942]
[robot_state_publisher-1] [INFO] [1746296709.617123559] [robot_state_publisher]: Robot initialized
[rviz2-3] [INFO] [1746296709.800654404] [rviz2]: Stereo is NOT SUPPORTED
[... 35 lines cut ...]
[INFO] [robot_state_publisher-1]: process started with pid [14118]
[INFO] [joint_state_publisher_gui-2]: process started with pid [14119]
[INFO] [rviz2-3]: process started with pid [14120]
[robot_state_publisher-1] [INFO] [1746296725.406869642] [robot_state_publisher]: Robot initialized
[rviz2-3] [INFO] [1746296725.608475738] [rviz2]: Stereo is NOT SUPPORTED
[rviz2-3] [INFO] [1746296725.608541025] [rviz2]: OpenGl version: 4.6 (GLSL 4.6)
[rviz2-3] [INFO] [1746296725.623588039] [rviz2]: Stereo is NOT SUPPORTED
[joint_state_publisher_gui-2] [INFO] [1746296725.754667916] [joint_state_publisher]: Waiting for robot_description to be published on the robot_description topic...
[joint_state_publisher_gui-2] [INFO] [1746296725.760565724] [joint_state_publisher]: Centering
[joint_state_publisher_gui-2] [INFO] [1746296725.907665971] [joint_state_publisher]: Centering
[rviz2-3] [ERROR] [1746296728.732680927] [rviz.ros_resource_retriever]: Timeout: Not able to call the service /rviz/get_resource
[... cut ...]
^C[WARNING] [launch]: user interrupted with ctrl-c (SIGINT)
[rviz2-3] [INFO] [1746296771.655381546] [rclcpp]: signal_handler(signum=2)
[robot_state_publisher-1] [INFO] [1746296771.655585797] [rclcpp]: signal_handler(signum=2)
[ERROR] [joint_state_publisher_gui-2]: process has died [pid 14119, exit code -2, cmd '/opt/ros/kilted/lib/joint_state_publisher_gui/joint_state_publisher_gui --ros-args'].
[INFO] [rviz2-3]: process has finished cleanly [pid 14120]
[INFO] [robot_state_publisher-1]: process has finished cleanly [pid 14118]
```

ahcorde, 2025-05-05:
Thank you for testing

closed 2025-05-05 by ahcorde