ros2/kilted_tutorial_party #110: Creating a content filtering subscription

tfoote, 2025-05-01:
Check the documentation for the 'Creating a content filtering subscription' page

## Setup
- Rmw: Fastdds
- BuildType: Debian
- Os: Noble
- Chip: Amd64

## Links
- [Creating a content filtering subscription page](https://docs.ros.org/en/kilted/Tutorials/Demos/Content-Filtering-Subscription.html)

## Checks
- [ ] **I was able to follow the documentation.**
- [ ] **The documentation seemed clear to me.**
- [ ] **The documentation didn't have any obvious errors.**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

avanmalleghem, 2025-05-03:
Interested in !

avanmalleghem, 2025-05-03:
PR : https://github.com/ros2/ros2_documentation/pull/5500

christophebedard, 2025-05-03:
@avanmalleghem other than the fixes in that PR, is the tutorial correct?

avanmalleghem, 2025-05-03:
@christophebedard yes it is :)

christophebedard, 2025-05-03:
Great, thanks for testing!

closed 2025-05-03 by christophebedard