ros2/kilted_tutorial_party #111: Creating A Workspace

tfoote, 2025-05-01:
Check the documentation for the 'Creating A Workspace' page

## Setup
- Rmw: Fastdds
- BuildType: Debian
- Os: Noble
- Chip: Amd64

## Links
- [Creating A Workspace page](https://docs.ros.org/en/kilted/Tutorials/Beginner-Client-Libraries/Creating-A-Workspace/Creating-A-Workspace.html)

## Checks
- [ ] **I was able to follow the documentation.**
- [ ] **The documentation seemed clear to me.**
- [ ] **The documentation didn't have any obvious errors.**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

JatinVira, 2025-05-13:
Hey @ahcorde 

I independently tried this out on the latest docker image

My setup
```
   PLATFORM INFORMATION
system           : Linux
platform info    : Linux-5.15.0-139-generic-x86_64-with-glibc2.39
release          : 5.15.0-139-generic
processor        : x86_64

   QOS COMPATIBILITY LIST
compatibility status    : No publisher/subscriber pairs found

   RMW MIDDLEWARE
middleware name    : rmw_fastrtps_cpp

   ROS 2 INFORMATION
distribution name      : kilted
distribution type      : ros2
distribution status    : pre-release
release platforms      : {'debian': ['bookworm'], 'rhel': ['9'], 'ubuntu': ['noble']}

   TOPIC LIST
topic               : none
publisher count     : 0
subscriber count    : 0

```

[image]

I **did not** see any such warning.

Here's a snippet of the same...

```
root@c27f868f0ac8:~/ros2_ws# ros2 run turtlesim turtlesim_node
QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/tmp/runtime-root'
[INFO] [1747151126.078235225] [turtlesim]: Starting turtlesim with node name /turtlesim
[INFO] [1747151126.083287049] [turtlesim]: Spawning turtle [turtle1] at x=[5.544445], y=[5.544445], theta=[0.000000]
_amdgpu_device_initialize: amdgpu_get_auth (1) failed (-1)
amdgpu: amdgpu_device_initialize failed.
glx: failed to create dri3 screen
failed to load driver: radeonsi
failed to open /dev/dri/card0: No such file or directory
failed to load driver: radeonsi
_amdgpu_device_initialize: amdgpu_get_auth (1) failed (-1)
amdgpu: amdgpu_device_initialize failed.
glx: failed to create dri3 screen
failed to load driver: radeonsi
failed to open /dev/dri/card0: No such file or directory
failed to load driver: radeonsi
^C[INFO] [1747151177.625141890] [rclcpp]: signal_handler(signum=2)
[ros2run]: Received signal:  Interrupt
[INFO] [1747151177.625338640] [rclcpp]: signal_handler(signum=2)
```

The warning's here are due to forwarding the video feed to my `xhost`

Sumary:

- I was able to follow the documentation.
- The documentation seemed clear to me.
- The documentation didn't have any obvious errors.

@ahcorde , we can successfully mark this issue completed and close it.

ahcorde, 2025-05-13:
Thank you for testing, 

This warning might appear if you have installed the package from debs. (I guess)

closed 2025-05-13 by ahcorde