ros2/kilted_tutorial_party #112: Creating an Action

tfoote, 2025-05-01:
Check the documentation for the 'Creating an Action' page

## Setup
- Rmw: Fastdds
- BuildType: Debian
- Os: Noble
- Chip: Amd64

## Links
- [Creating an Action page](https://docs.ros.org/en/kilted/Tutorials/Intermediate/Creating-an-Action.html)

## Checks
- [ ] **I was able to follow the documentation.**
- [ ] **The documentation seemed clear to me.**
- [ ] **The documentation didn't have any obvious errors.**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

SuperGops7, 2025-05-03:
@tfoote @ahcorde I'd like to test this. Can you assign this to me?

SuperGops7, 2025-05-04:
- [x] I was able to follow the documentation.
- [x] The documentation seemed clear to me.
- [ ] The documentation didn't have any obvious errors.

Although the documentation does work, I feel there is an error in the command to source our workspace for linux, as shown in the screenshot of the docs page attached. Shouldn't it be `source install/local_setup.bash`?

[image]

[image]

SuperGops7, 2025-05-04:
I have raised an issue in the official ros2 docs repo with the same comment here: [ROS2 docs Issue](https://github.com/ros2/ros2_documentation/issues/5527)

christophebedard, 2025-05-04:
Thanks for testing. Since you opened an issue in the docs repo and the tutorial otherwise works, I'll close this.

closed 2025-05-04 by christophebedard