ros2/kilted_tutorial_party #113: Creating Your First ROS2 Package

tfoote, 2025-05-01:
Check the documentation for the 'Creating Your First ROS2 Package' page

## Setup
- Rmw: Fastdds
- BuildType: Debian
- Os: Noble
- Chip: Amd64

## Links
- [Creating Your First ROS2 Package page](https://docs.ros.org/en/kilted/Tutorials/Beginner-Client-Libraries/Creating-Your-First-ROS2-Package.html)

## Checks
- [ ] **I was able to follow the documentation.**
- [ ] **The documentation seemed clear to me.**
- [ ] **The documentation didn't have any obvious errors.**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

JatinVira, 2025-05-13:
Hey,
I independently tried this out using the latest docker image

- I was able to follow the documentation 
- The documentation seemed clear to me
- The documentation didn't have any obvious errors

Here's some terminal logs
```
root@11038f293c80:/home/ubuntu# cd ros2_ws/src/
root@11038f293c80:/home/ubuntu/ros2_ws/src# ros2 pkg create --build-type ament_cmake --license Apache-2.0 testing_pkg
going to create a new package
package name: testing_pkg
destination directory: /home/ubuntu/ros2_ws/src
package format: 3
version: 0.0.0
description: TODO: Package description
maintainer: ['root <root@todo.todo>']
licenses: ['Apache-2.0']
build type: ament_cmake
dependencies: []
creating folder ./testing_pkg
creating ./testing_pkg/package.xml
creating source and include folder
creating folder ./testing_pkg/src
creating folder ./testing_pkg/include/testing_pkg
creating ./testing_pkg/CMakeLists.txt
root@11038f293c80:/home/ubuntu/ros2_ws/src# ros2 pkg create --build-type ament_cmake --license Apache-2.0 --node-name my_node my_package
going to create a new package
package name: my_package
destination directory: /home/ubuntu/ros2_ws/src
package format: 3
version: 0.0.0
description: TODO: Package description
maintainer: ['root <root@todo.todo>']
licenses: ['Apache-2.0']
build type: ament_cmake
dependencies: []
node_name: my_node
creating folder ./my_package
creating ./my_package/package.xml
creating source and include folder
creating folder ./my_package/src
creating folder ./my_package/include/my_package
creating ./my_package/CMakeLists.txt
creating ./my_package/src/my_node.cpp
root@11038f293c80:/home/ubuntu/ros2_ws/src# cd ..
root@11038f293c80:/home/ubuntu/ros2_ws# colcon build
Starting >>> my_package
Starting >>> testing_pkg
Finished <<< testing_pkg [0.94s]                                                      
Finished <<< my_package [1.02s]                  

Summary: 2 packages finished [1.10s]
root@11038f293c80:/home/ubuntu/ros2_ws# source install/setup.bash 
root@11038f293c80:/home/ubuntu/ros2_ws# ros2 run my_package my_node
hello world my_package package
root@11038f293c80:/home/ubuntu/ros2_ws# 

```

here's my setup

```
   PLATFORM INFORMATION
system           : Linux
platform info    : Linux-5.15.0-139-generic-x86_64-with-glibc2.39
release          : 5.15.0-139-generic
processor        : x86_64

   QOS COMPATIBILITY LIST
compatibility status    : No publisher/subscriber pairs found

   RMW MIDDLEWARE
middleware name    : rmw_fastrtps_cpp

   ROS 2 INFORMATION
distribution name      : kilted
distribution type      : ros2
distribution status    : pre-release
release platforms      : {'debian': ['bookworm'], 'rhel': ['9'], 'ubuntu': ['noble']}
```

@ahcorde 

This has been verified by two people.
Please assign this issue to @jmackay2 and to me.

Thankyou.

JatinVira, 2025-05-13:
We can successfully close this issue

ahcorde, 2025-05-13:
Thank you for testing

closed 2025-05-13 by ahcorde