ros2/kilted_tutorial_party #114: Custom ROS2 Interfaces

tfoote, 2025-05-01:
Check the documentation for the 'Custom ROS2 Interfaces' page

## Setup
- Rmw: Fastdds
- BuildType: Debian
- Os: Noble
- Chip: Amd64

## Links
- [Custom ROS2 Interfaces page](https://docs.ros.org/en/kilted/Tutorials/Beginner-Client-Libraries/Custom-ROS2-Interfaces.html)

## Checks
- [ ] **I was able to follow the documentation.**
- [ ] **The documentation seemed clear to me.**
- [ ] **The documentation didn't have any obvious errors.**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

dheerubhai-101, 2025-05-03:
Testing this for C++

- [x] I was able to follow the documentation.
- [x] The documentation seemed clear to me.
- [ ] The documentation didn't have any obvious errors.

Msg and Srv creation confirmed

[image]

Testing new interfaces for pub/sub

[image]

Testing new interfaces for service/client has two errors with the client code.
1. Code expects arguments when running `ros2 run cpp_srvcli client` but code block does not does not include that in documentation
2. Code has hardcoded values

So `client` exists with error code 1.
```
  if (argc != 4) { // CHANGE
      RCLCPP_INFO(rclcpp::get_logger("rclcpp"), "usage: add_three_ints_client X Y Z");      // CHANGE
      return 1;
  }

  std::shared_ptr<rclcpp::Node> node = rclcpp::Node::make_shared("add_three_ints_client");  // CHANGE
  rclcpp::Client<tutorial_interfaces::srv::AddThreeInts>::SharedPtr client =                // CHANGE
    node->create_client<tutorial_interfaces::srv::AddThreeInts>("add_three_ints");          // CHANGE

  auto request = std::make_shared<tutorial_interfaces::srv::AddThreeInts::Request>();       // CHANGE
  request->a = 41;
  request->b = 1;
  request->c = 1;                                                                           // CHANGE

```

Changing this to the version similar to Jazzy documentation

christophebedard, 2025-05-04:
Thanks for testing. The fix is in https://github.com/ros2/ros2_documentation/pull/5528.

closed 2025-05-04 by christophebedard