ros2/kilted_tutorial_party #116: Debugging Tf2 Problems

tfoote, 2025-05-01:
Check the documentation for the 'Debugging Tf2 Problems' page

## Setup
- Rmw: Fastdds
- BuildType: Debian
- Os: Noble
- Chip: Amd64

## Links
- [Debugging Tf2 Problems page](https://docs.ros.org/en/kilted/Tutorials/Intermediate/Tf2/Debugging-Tf2-Problems.html)

## Checks
- [ ] **I was able to follow the documentation.**
- [ ] **The documentation seemed clear to me.**
- [ ] **The documentation didn't have any obvious errors.**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

avanmalleghem, 2025-05-03:
Interested in !

avanmalleghem, 2025-05-03:
* [x]  **I was able to follow the documentation.**
* [x]  **The documentation seemed clear to me.**
* [x]  **The documentation didn't have any obvious errors.**

Can be closed 👍

ahcorde, 2025-05-05:
Thank you for testing

closed 2025-05-05 by ahcorde