ros2/kilted_tutorial_party #117: Demo Nodes (C++) executables

tfoote, 2025-05-01:
## Setup
- Rmw: Cyclone
- BuildType: Binary
- Os: Windows
- Chip: Amd64

## Links
- [Demo Nodes C++](https://github.com/ros2/demos/blob/kilted/demo_nodes_cpp/README.md)

## Checks
- [ ] **The executables in the README are present and work as expected.**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

Poo-09, 2025-05-03:
**The executables in the README are present and work as expected.**
_ros2 run demo_nodes_cpp talker,ros2 run demo_nodes_cpp listener_
[image]

_ros2 run demo_nodes_cpp listener_best_effort_
[image]

Basic Server & Client[Synchronous]
[image]
Client [Asynchronous]

[image]

One-OffTimer

[image]

Reuse Timer

[image]

Poo-09, 2025-05-03:
**Serialized Listening**

[image]

**Content-Filter Messaging**

[image]

**List Parameters**
Synchronous

[image]

Asynchronous

[image]

**Set & Get Parameters**
Synchronous

[image]

Asynchronous

[image]

**Allocator Tutorial**

[image]

**Parameter Events**
Synchronous

[image]

Asynchronous

[image]

**Even Parameters Node**

[image]

**Set Parameters Callback**

[image]

**Parameter Blackboard**

[image]

**Parameter Event Handler**

[image]

**Loaned Messager Talker**

[image]

**Matched Event Detect**

[image]

ahcorde, 2025-05-05:
Thank you for testing On Windows! :+1:

closed 2025-05-05 by ahcorde