ros2/kilted_tutorial_party #118: Demo Nodes (C++) executables

tfoote, 2025-05-01:
## Setup
- Rmw: Cyclone
- BuildType: Debian
- Os: Noble
- Chip: Amd64

## Links
- [Demo Nodes C++](https://github.com/ros2/demos/blob/kilted/demo_nodes_cpp/README.md)

## Checks
- [ ] **The executables in the README are present and work as expected.**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

suchetanrs, 2025-05-03:
I am working on this

suchetanrs, 2025-05-03:
I had a few issues:

1. https://github.com/ros2/demos/blob/kilted/demo_nodes_cpp/README.md#asynchronous-2 the command uses even_parameters_node but it should be even_parameter_node

2. Although working, the values below don't match the documentation for Set Parameters Callback

```
root@cluttered:~# ros2 param set /even_parameters_node myint 2
Set parameter successful
root@cluttered:~# ros2 param set /even_parameters_node myint 3
Setting parameter failed: only even integers can be set
root@cluttered:~# ros2 param get /set_param_callback_node param1
Double value is: 1.0
root@cluttered:~# ros2 param get /set_param_callback_node param2
Double value is: 2.0
root@cluttered:~# ros2 param set set_param_callback_node param1 10.0
Setting parameter failed: cannot set 'param1' > 5.0
root@cluttered:~# ros2 param set set_param_callback_node param1 3.0
Set parameter successful
root@cluttered:~# ros2 param get /set_param_callback_node param1 
Double value is: 3.0
root@cluttered:~# ros2 param get /set_param_callback_node param2
Double value is: 4.0
root@cluttered:~# 
root@cluttered:~# 
```

3. I verified my rmw implementation through multiple ways and it was cyclonedds. However, I got
```
launching: talker_loaned_message

Press…[INFO] [1746244595.653332530] [rclcpp]: Currently used middleware can't loan messages. Local allocator will be used.
[INFO] [1746244595.653405347] [loaned_message_talker]: Publishing: '1.000000'
[INFO] [1746244595.653452306] [rclcpp]: Currently used middleware can't loan messages. Local allocator will be used.
[INFO] [1746244595.653463788] [loaned_message_talker]: Publishing: 'Hello World: 1'
[INFO] [1746244596.653296818] [loaned_message_talker]: Publishing: '2.000000'
[INFO] [1746244596.653354477] [loaned_message_talker]: Publishing: 'Hello World: 2'
```

4. I also got a warning when performing content filtering

`Press…[WARN] [1746244381.084542445] [content_filtering_subscriber]: Content filter is not enabled since it's not supported`

Script used can be found here https://github.com/suchetanrs/kilted_tutorial_party/tree/main/scripts

Full log output is below:
```
root@cluttered:~# source /opt/ros/kilted/setup.bash
root@cluttered:~# export RMW_IMPLEMENTATION=rmw_cyclonedds_cpp
root@cluttered:~# ros2 doctor --report | grep middleware
middleware name    : rmw_cyclonedds_cpp
root@cluttered:~# uname -m
x86_64
root@cluttered:~# ./scripts/demo_nodes_cpp_executables.sh 

==================================== Group: Basic Talker & Listener =======================================
===========================================================================
===========================================================================
===========================================================================
===========================================================================
launching: talker
launching: listener
launching: listener_best_effort

Press…[INFO] [1746244328.679110474] [talker]: Publishing: 'Hello World: 1'
[INFO] [1746244328.679439974] [listener]: I heard: [Hello World: 1]
[INFO] [1746244328.679462157] [listener]: I heard: [Hello World: 1]
[INFO] [1746244329.679137077] [talker]: Publishing: 'Hello World: 2'
[INFO] [1746244329.679420749] [listener]: I heard: [Hello World: 2]
[INFO] [1746244329.679473644] [listener]: I heard: [Hello World: 2]
[... 600 lines cut ...]
Double value is: 4.0
root@cluttered:~# 
root@cluttered:~# 
root@cluttered:~# 
root@cluttered:~# 
root@cluttered:~# 
root@cluttered:~# ros2 param list
/parameter_blackboard:
  qos_overrides./parameter_events.publisher.depth
  qos_overrides./parameter_events.publisher.durability
  qos_overrides./parameter_events.publisher.history
  qos_overrides./parameter_events.publisher.reliability
  start_type_description_service
  use_sim_time
root@cluttered:~# ros2 param set this_node an_int_param 21
Set parameter successful
root@cluttered:~# 
root@cluttered:~# 
root@cluttered:~# 
root@cluttered:~# 
root@cluttered:~# ros2 param set /a_namespace/a_remote_node a_string_param "string value to set"
Set parameter successful
root@cluttered:~# 

```

ahcorde, 2025-05-13:
Thank you for testing, PR was merged https://github.com/ros2/demos/pull/715

closed 2025-05-13 by ahcorde