ros2/kilted_tutorial_party #119: Demo Nodes (C++) executables

tfoote, 2025-05-01:
## Setup
- Rmw: Fastdds
- BuildType: Binary
- Os: Windows
- Chip: Amd64

## Links
- [Demo Nodes C++](https://github.com/ros2/demos/blob/kilted/demo_nodes_cpp/README.md)

## Checks
- [ ] **The executables in the README are present and work as expected.**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

ShahazadAbdulla, 2025-05-04:
I'll test this on Windows 11 / amd64 / binary. Will report results for both Fast DDS and CycloneDDS where applicable, noting the RMW conflict in the setup vs. Loaned Message description

Tanishq30052002, 2025-05-13:
1. `add_two_ints_client` & `add_two_ints_client_async` & `add_two_ints_server`
[image]

2. `listener_serialized_message`  & `talker_serialized_message`
[image]

3. `reuse_timer` & `one_off_timer`
[image]

4. `content_filtering_subscriber` & `content_filtering_publisher`
[image]

5. `list_parameters` & `list_parameters_async` & `set_and_get_parameters` & `set_and_get_parameters_async`
[image]

6. `allocator_tutorial`
[image]

7. `parameter_events` & `parameter_events_async`
[image]

8. `parameter_event_handler`
[image]

9. `talker_loaned_message`
[image]

10. `matched_event_detect`
[image]

11. `even_parameters_node` & `set_parameters_callback` & `parameter_blackboard`
[image]

ahcorde, 2025-05-14:
Thank you for testing

closed 2025-05-14 by ahcorde