ros2/kilted_tutorial_party #12: Building ROS2 Package with eclipse 2021 06

tfoote, 2025-05-01:
Check the documentation for the 'Building ROS2 Package with eclipse 2021 06' page

## Setup
- Rmw: Fastdds
- BuildType: Debian
- Os: Noble
- Chip: Amd64

## Links
- [Building ROS2 Package with eclipse 2021 06 page](https://docs.ros.org/en/kilted/Tutorials/Miscellaneous/Building-ROS2-Package-with-eclipse-2021-06.html)

## Checks
- [ ] **I was able to follow the documentation.**
- [ ] **The documentation seemed clear to me.**
- [ ] **The documentation didn't have any obvious errors.**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

Aniketkumarroy, 2025-05-06:
Hi I will check this

Aniketkumarroy, 2025-05-09:
Hi I tested this, its working fine

https://github.com/user-attachments/assets/210024bf-b81f-4b17-b12d-da96990b6d9b

- [x] I was able to follow the documentation.
- [ ] The documentation seemed clear to me.
- [x] The documentation didn't have any obvious errors.

the instructions given are not complete and miss some pictures which create great confusion for new users of eclipse to follow. also the library path hasn't been updated from /opt/ros/iron to /opt/ros/kilted
should I raise a PR to fix this

Yadunund, 2025-05-09:
> the instructions given are not complete and miss some pictures which create great confusion for new users of eclipse to follow. also the library path hasn't been updated from /opt/ros/iron to /opt/ros/kilted
should I raise a PR to fix this @mjcarroll

@Aniketkumarroy yes please open a PR to https://github.com/ros2/ros2_documentation targeting the `rolling` branch and we will backport the fix to older distros. 

Closing this ticket out! thanks for testing.

closed 2025-05-09 by Yadunund