ros2/kilted_tutorial_party #120: Demo Nodes (C++) executables

tfoote, 2025-05-01:
## Setup
- Rmw: Fastdds
- BuildType: Debian
- Os: Noble
- Chip: Amd64

## Links
- [Demo Nodes C++](https://github.com/ros2/demos/blob/kilted/demo_nodes_cpp/README.md)

## Checks
- [ ] **The executables in the README are present and work as expected.**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

suchetanrs, 2025-05-03:
I am working on this

suchetanrs, 2025-05-03:
I had a few issues:

1. https://github.com/ros2/demos/blob/kilted/demo_nodes_cpp/README.md#asynchronous-2 the command uses even_parameters_node but it should be even_parameter_node
2. Although working, the values below don't match the documentation for `Set Parameters Callback`
3. Loaned messages are not there in fastdds so I get an error. I think it is OK.

Everything else works fine.

```
root@cluttered:~# ros2 param set /even_parameters_node myint 2
Set parameter successful
root@cluttered:~# ros2 param set /even_parameters_node myint 3
Setting parameter failed: only even integers can be set
root@cluttered:~# ros2 param get /set_param_callback_node param1
Double value is: 1.0
root@cluttered:~# ros2 param get /set_param_callback_node param2
Double value is: 2.0
root@cluttered:~# ros2 param set set_param_callback_node param1 10.0
Setting parameter failed: cannot set 'param1' > 5.0
root@cluttered:~# ros2 param set set_param_callback_node param1 3.0
Set parameter successful
root@cluttered:~# ros2 param get /set_param_callback_node param1 
Double value is: 3.0
root@cluttered:~# ros2 param get /set_param_callback_node param2
Double value is: 4.0
root@cluttered:~# 
root@cluttered:~# 
```

Script used can be found here https://github.com/suchetanrs/kilted_tutorial_party/tree/main/scripts

Full log output below.
```
root@cluttered:~# ros2 node list
/rqt_gui_py_node_1519732
root@cluttered:~# source /opt/ros/kilted/setup.bash
root@cluttered:~# ros2 doctor --report | grep middleware
middleware name    : rmw_fastrtps_cpp
root@cluttered:~# ./scripts/demo_nodes_cpp_executables.sh 

==================================== Group: Basic Talker & Listener =======================================
===========================================================================
===========================================================================
===========================================================================
===========================================================================
launching: talker
launching: listener
launching: listener_best_effort

Press…[INFO] [1746243697.539558749] [talker]: Publishing: 'Hello World: 1'
[INFO] [1746243697.539909811] [listener]: I heard: [Hello World: 1]
[INFO] [1746243697.539909828] [listener]: I heard: [Hello World: 1]
[INFO] [1746243698.539596983] [talker]: Publishing: 'Hello World: 2'
[INFO] [1746243698.539916848] [listener]: I heard: [Hello World: 2]
[INFO] [1746243698.539945644] [listener]: I heard: [Hello World: 2]
[INFO] [1746243699.539590392] [talker]: Publishing: 'Hello World: 3'
[INFO] [1746243699.539898890] [listener]: I heard: [Hello World: 3]
[INFO] [1746243699.539930688] [listener]: I heard: [Hello World: 3]

[ros2run]: Received signal:  Terminated
[ros2run]: Received signal:  Terminated
[INFO] [1746243700.008095545] [rclcpp]: signal_handler(signum=15)
[ros2run]: Received signal:  Terminated
[INFO] [1746243700.008146168] [rclcpp]: signal_handler(signum=15)
[INFO] [1746243700.008154130] [rclcpp]: signal_handler(signum=15)

==================================== Group: Basic Server & Clients =======================================
===========================================================================
===========================================================================
===========================================================================
[... 536 lines cut ...]
root@cluttered:~# 
root@cluttered:~# 
root@cluttered:~# 
root@cluttered:~# ros2 param list
/parameter_blackboard:
  qos_overrides./parameter_events.publisher.depth
  qos_overrides./parameter_events.publisher.durability
  qos_overrides./parameter_events.publisher.history
  qos_overrides./parameter_events.publisher.reliability
  start_type_description_service
  use_sim_time
/rqt_gui_py_node_1519732:
  start_type_description_service
  use_sim_time
root@cluttered:~# 
root@cluttered:~# 
root@cluttered:~# 
root@cluttered:~# 
root@cluttered:~# 
root@cluttered:~# ros2 param set this_node an_int_param 21
Set parameter successful
root@cluttered:~# ros2 param set /a_namespace/a_remote_node a_string_param "string value to set"
Set parameter successful
root@cluttered:~# 
```

ahcorde, 2025-05-13:
Thank you for testing, related PR with the fix was merged

closed 2025-05-13 by ahcorde