ros2/kilted_tutorial_party #122: Demo Nodes Native (C++) executables

tfoote, 2025-05-01:
## Setup
- Rmw: Cyclone
- BuildType: Binary
- Os: Windows
- Chip: Amd64

## Links
- [Demo Nodes C++ native](https://github.com/ros2/demos/blob/kilted/demo_nodes_cpp_native/README.md)

## Checks
- [ ] **The executables in the README are present and work as expected.**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

ShahazadAbdulla, 2025-05-04:
Anyway I'll test this on Windows 11 / amd64 / binary / Fast DDS (pending clarification on RMW)

Aniketkumarroy, 2025-05-14:
Hi I tested this
### **Setup**

[image]
### **Running Demo Node Native**

[image]
again checking `ros2 doctor --report` to verify that it is cyclone DDS

[image]

closed 2025-05-16 by mjcarroll