ros2/kilted_tutorial_party #124: Demo Nodes Native (C++) executables

tfoote, 2025-05-01:
## Setup
- Rmw: Fastdds
- BuildType: Binary
- Os: Windows
- Chip: Amd64

## Links
- [Demo Nodes C++ native](https://github.com/ros2/demos/blob/kilted/demo_nodes_cpp_native/README.md)

## Checks
- [ ] **The executables in the README are present and work as expected.**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

pradyum, 2025-05-03:
**The executable is working fine**

---

```
C:\pixi_ws>ros2 run demo_nodes_cpp_native talker
[INFO] [1746230573.405225500] [talker_native]: eprosima::fastdds::dds::DomainParticipant * 2572758432496
[INFO] [1746230573.415353200] [talker_native]: eprosima::fastdds::dds::DataWriter * 2572794539376
[INFO] [1746230573.922941100] [talker_native]: Publishing: 'Hello World: 1'
[INFO] [1746230574.423215200] [talker_native]: Publishing: 'Hello World: 2'
[INFO] [1746230574.923773600] [talker_native]: Publishing: 'Hello World: 3'
[INFO] [1746230575.422822300] [talker_native]: Publishing: 'Hello World: 4'
[INFO] [1746230575.923843300] [talker_native]: Publishing: 'Hello World: 5'
[INFO] [1746230576.423954300] [talker_native]: Publishing: 'Hello World: 6'
[INFO] [1746230578.736533400] [talker_native]: Publishing: 'Hello World: 7'
[INFO] [1746230578.923130800] [talker_native]: Publishing: 'Hello World: 8'
[INFO] [1746230579.423056300] [talker_native]: Publishing: 'Hello World: 9'
[INFO] [1746230579.922744000] [talker_native]: Publishing: 'Hello World: 10'
```

ahcorde, 2025-05-05:
Thank you for testing!

closed 2025-05-05 by ahcorde