ros2/kilted_tutorial_party #125: Demo Nodes Native (C++) executables

tfoote, 2025-05-01:
## Setup
- Rmw: Fastdds
- BuildType: Debian
- Os: Noble
- Chip: Amd64

## Links
- [Demo Nodes C++ native](https://github.com/ros2/demos/blob/kilted/demo_nodes_cpp_native/README.md)

## Checks
- [ ] **The executables in the README are present and work as expected.**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

suchetanrs, 2025-05-03:
Works

```
root@cluttered:~# source /opt/ros/kilted/setup.bash
root@cluttered:~# source /opt/ros/kilted/setup.bash
root@cluttered:~# ros2 run demo_nodes_cpp_native talker
[INFO] [1746245206.128876086] [talker_native]: eprosima::fastdds::dds::DomainParticipant * 106725476165440
[INFO] [1746245206.129379924] [talker_native]: eprosima::fastdds::dds::DataWriter * 106725479158864
[INFO] [1746245206.629027702] [talker_native]: Publishing: 'Hello World: 1'
[INFO] [1746245207.129136102] [talker_native]: Publishing: 'Hello World: 2'
[INFO] [1746245207.629028100] [talker_native]: Publishing: 'Hello World: 3'
[INFO] [1746245208.129128817] [talker_native]: Publishing: 'Hello World: 4'
^C[INFO] [1746245208.212744769] [rclcpp]: signal_handler(signum=2)
[ros2run]: Received signal:  Interrupt
[INFO] [1746245208.212811150] [rclcpp]: signal_handler(signum=2)
root@cluttered:~# ros2 doctor --report | grep middleware
middleware name    : rmw_fastrtps_cpp
root@cluttered:~# uname -m
x86_64
root@cluttered:~# 
```

ahcorde, 2025-05-05:
thank you for testing

closed 2025-05-05 by ahcorde