ros2/kilted_tutorial_party #127: Demos

tfoote, 2025-05-01:
Check the documentation for the 'Demos' page

## Setup
- Rmw: Fastdds
- BuildType: Debian
- Os: Noble
- Chip: Amd64

## Links
- [Demos page](https://docs.ros.org/en/kilted/Tutorials/Demos.html)

## Checks
- [ ] **I was able to follow the documentation.**
- [ ] **The documentation seemed clear to me.**
- [ ] **The documentation didn't have any obvious errors.**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

Tenveis, 2025-05-03:
I am interested.

Tenveis, 2025-05-03:
Following are the results:
- [x] I was able to follow the documentation.
- [x] The documentation seemed clear to me.
- [x] The documentation didn't have any obvious errors.

All external links are working fine. 
[image]

Do I need to test each external link page? please let me know. Thank you.

ahcorde, 2025-05-05:
Thank you for testing