ros2/kilted_tutorial_party #129: Discovery Server

tfoote, 2025-05-01:
Check the documentation for the 'Discovery Server' page

## Setup
- Rmw: Fastdds
- BuildType: Debian
- Os: Noble
- Chip: Amd64

## Links
- [Discovery Server page](https://docs.ros.org/en/kilted/Tutorials/Advanced/Discovery-Server/Discovery-Server.html)

## Checks
- [ ] **I was able to follow the documentation.**
- [ ] **The documentation seemed clear to me.**
- [ ] **The documentation didn't have any obvious errors.**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

virkhanna20, 2025-05-04:
I will do this .

danielcranston, 2025-05-04:
- [x] **I was able to follow the documentation.**
- [x] **The documentation seemed clear to me.**
- [x] **The documentation didn't have any obvious errors.**

Some small nitpicks

* "It is also possible to configure the discovery server using the fastdds tool and an environment variable [as opposed to XML configuration], which is the approach used in this tutorial". But towards the end, the tutorial also uses XML configuration though.
* "using the Discovery Server v2 these nodes will not have all the network information, and thus their functionality will be limited." sentence a bit long
* "Most of these [CLI] tools use DDS simple discovery to exchange topic information with every existing participant". This is a bit surprising to me, it sounds like the temporary nodes that CLI verbs create are hard-wired to use DDS simple discovery whether you like it or not. But there's nothing stopping you from setting `ROS_DISCOVERY_SERVER` for these as well.
* Super Client explanations could probably be made more succinct.
* `--remap __node:=listener` is redundant in Super Client example (same for talker). Also the case for "No Daemon" example.
* Are comments saying "Use Foxy or later" really needed at this point? That's pretty far back at this point.

I'll try running all the steps soon, will update. Specifically I'm a bit surprised that some things mention the new name (Fastdds) and some the "old" e.g.`FASTRTPS_DEFAULT_PROFILES_FILE`

danielcranston, 2025-05-04:
@virkhanna20 damn sorry I should have claimed this first. Feel free to do it as well.

danielcranston, 2025-05-04:
I said I'd run the steps, so here goes.

### Invalid arguments
In the "Server Redundancy" section, the command given is:

> fastdds discovery --server-id 0 --ip-address 127.0.0.1 --port 11811

But  `--ip-address` and `--port` are not longer valid arguments. Seems these are called `--udp-address` (`-l`) and `--udp-port` (`-p`) these days.

### Discovery partitions test

The "Discovery partitions" test worked as expected.

### Deprecated environment variable
In the "Daemon's related tools" section, I found out that indeed `FASTRTPS_DEFAULT_PROFILES_FILE` is deprecated:

```
[WARN] [1746370021.254660133] [rmw_fastrtps_shared_cpp]: FASTRTPS_DEFAULT_PROFILES_FILE value is used for participant configuration, but it is deprecated and will no longer be supported. Use FASTDDS_DEFAULT_PROFILES_FILE instead.
```

### Error when using --no-daemon
In "No daemon tools" section, I ran into an issue:

```
$ export FASTDDS_DEFAULT_PROFILES_FILE=super_client_configuration_file.xml
$ ros2 topic list --no-daemon
2025-05-04 23:55:03.390 [XMLPARSER Error] Node 'discoveryServersList' without content -> Function getXMLLocatorList
2025-05-04 23:55:03.391 [XMLPARSER Error] Error parsing participant profile -> Function parseXMLParticipantProf
2025-05-04 23:55:03.391 [XMLPARSER Error] Error parsing profile's tag participant -> Function parseProfiles
2025-05-04 23:55:03.391 [XMLPARSER Error] Error parsing '/home/daniel/super_client_configuration_file.xml' -> Function loadXMLFile
/parameter_events
/rosout
```

So I gather the structure of the XML file changed around the same time the "FastRTPS" -> "FASTDDS" change happened.

I'll make a PR to ros_documentation, but for this last one I'd appreciate some help figuring out the cause of the `--no-daemon` failure.

danielcranston, 2025-05-08:
The PR was merged, but the `--no-daemon` issue mentioned above is still unresolved.

danielcranston, 2025-05-10:
I'm guessing the pasted XML configuration profile structure is not up to date. Based on https://fast-dds.docs.eprosima.com/en/latest/fastdds/xml_configuration/domainparticipant.html#builtin-parameters I changed it like this:

```diff
<?xml version="1.0" encoding="UTF-8"?>
<dds>
    <profiles xmlns="http://www.eprosima.com/XMLSchemas/fastRTPS_Profiles">
        <participant profile_name="super_client_profile" is_default_profile="true">
            <rtps>
                <builtin>
                    <discovery_config>
                        <discoveryProtocol>SUPER_CLIENT</discoveryProtocol>
                        <discoveryServersList>
-                            <RemoteServer prefix="44.53.00.5f.45.50.52.4f.53.49.4d.41">
-                                <metatrafficUnicastLocatorList>
-                                    <locator>
-                                        <udpv4>
-                                            <address>127.0.0.1</address>
-                                            <port>11811</port>
-                                        </udpv4>
-                                    </locator>
-                                </metatrafficUnicastLocatorList>
-                            </RemoteServer>
+                            <locator>
+                                <udpv4>
+                                    <address>127.0.0.1</address>
+                                    <port>11811</port>
+                                </udpv4>
+                            </locator>
                        </discoveryServersList>
                    </discovery_config>
                </builtin>
            </rtps>
        </participant>
    </profiles>
</dds>
```

This fixes the `--no-daemon` issue.

Then I found https://fast-dds.docs.eprosima.com/en/latest/fastdds/ros2/discovery_server/ros2_discovery_server.html, which seems to be essentially the same article as the docs.ros.org one. I'm assuming it's more up to date. Maybe we should just link to it?

ahcorde, 2025-05-13:
Documentation was updated, Thank you for testing