ros2/kilted_tutorial_party #13: Check action Cli

tfoote, 2025-05-01:
## Setup
- Rmw: Cyclone
- BuildType: Binary
- Os: Windows
- Chip: Amd64

## Checks
- [ ] **Check info**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run examples_rclcpp_minimal_action_server action_server_member_functions
       ```
    1.
       ```bash
       # User input in terminal 2
       ros2 action info /fibonacci
       ```
  
    **Expect**
    
    1.
       ```bash
       # stdout in terminal 2
       Action: /fibonacci
       Action clients: 0
       Action servers: 1
         /minimal_action_server
       ```
  </details>
- [ ] **Check list**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run examples_rclcpp_minimal_action_server action_server_member_functions
       ```
    1.
       ```bash
       # User input in terminal 2
       ros2 action list
       ```
  
    **Expect**
    
    1.
       ```bash
       # stdout in terminal 2
       /fibonacci
       ```
  </details>
- [ ] **Check send_goal**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run examples_rclcpp_minimal_action_server action_server_member_functions
       ```
    1.
       ```bash
       # User input in terminal 2
       ros2 action send_goal /fibonacci example_interfaces/action/Fibonacci order:\ 2
       ```
  
    **Expect**
    
    1.
       ```bash
       # stdout in terminal 2
       Waiting for an action server to become available...
       Sending goal:
           order: 2
       
       Goal accepted with ID: 0110dbd59e804374bff8e4900c3201e8
       
       Result:
           sequence:
       - 0
       - 1
       - 1
       
       Goal finished with status: SUCCEEDED
       ```
  </details>

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

Poo-09, 2025-05-02:
[image]

[image]

[image]

First two checklists have the expected output, last one does not give the expected output.

ahcorde, 2025-05-02:
can try to investigate why this is happening ?

Maybe the command is wrong

ShahazadAbdulla, 2025-05-03:
Hi @Poo-09 and @ahcorde 

Helping out here I have encountered the same problem for the last 'check send_goal' and noticed the same issue.

I noticed in https://github.com/ros2/kilted_tutorial_party/issues/15#issuecomment-2847720326 that JasperGrant who suggested the dictionary as argument and that works.

```
(pixi_ros2_rolling) C:\pixi_ws>ros2 action send_goal /fibonacci example_interfaces/action/Fibonacci "{order: 2}"
Waiting for an action server to become available...
Sending goal:
     order: 2

Goal accepted with ID: 3757985db75647efa55900720339dc06

Result:
    sequence:
- 0
- 1
- 1

Goal finished with status: SUCCEEDED

(pixi_ros2_rolling) C:\pixi_ws>
```
this works

Poo-09, 2025-05-03:
[image]

> can try to investigate why this is happening ?
> 
> Maybe the command is wrong

yes the command was wrong
Thanks @ShahazadAbdulla for jumping in and confirming the issue. I just tested the dictionary-style input as you suggested. And it does work correctly, returning the expected sequence.

[image]

mjcarroll, 2025-05-09:
https://github.com/osrf/ros2_test_cases/issues/1696