ros2/kilted_tutorial_party #130: Dummy Map Server executables

tfoote, 2025-05-01:
## Setup
- Rmw: Cyclone
- BuildType: Binary
- Os: Windows
- Chip: Amd64

## Links
- [Dummy Map Server](https://github.com/ros2/demos/blob/kilted/dummy_robot/dummy_map_server/README.md#run)

## Checks
- [ ] **The executables in the README are present and work as expected.**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

ShahazadAbdulla, 2025-05-04:
Successfully completed the check for Issue #130 on Windows 11 / amd64 / binary / CycloneDDS.

- Ran `ros2 run dummy_map_server dummy_map_server` in Terminal 1.
- Ran `ros2 topic echo /map` in Terminal 2.
- An `nav_msgs/msg/OccupancyGrid` message with the expected structure (header, info, data) was successfully received and echoed in Terminal 2.

The `dummy_map_server` executable started and published data as expected.

ahcorde, 2025-05-05:
Thank you for testing

closed 2025-05-05 by ahcorde