ros2/kilted_tutorial_party #132: Dummy Map Server executables

tfoote, 2025-05-01:
## Setup
- Rmw: Fastdds
- BuildType: Binary
- Os: Windows
- Chip: Amd64

## Links
- [Dummy Map Server](https://github.com/ros2/demos/blob/kilted/dummy_robot/dummy_map_server/README.md#run)

## Checks
- [ ] **The executables in the README are present and work as expected.**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

pradyum, 2025-05-03:
**The dummy map server is working fine**

---
Node
```
C:\pixi_ws>ros2 run dummy_map_server dummy_map_server
```

Output
```
C:\pixi_ws>ros2 topic echo /map --once
header:
  stamp:
    sec: 1746230426
    nanosec: 500717500
  frame_id: world
info:
  map_load_time:
    sec: 0
    nanosec: 0
  resolution: 0.10000000149011612
  width: 100
  height: 100
  origin:
    position:
      x: -5.0
      y: -5.0
      z: 0.0
    orientation:
      x: 0.0
      y: 0.0
      z: 0.0
      w: 1.0
data:
- -1
- -1
- -1
- -1
- -1
- -1
- -1
- -1
- -1
- -1
- -1
- -1
- -1
- -1
- -1
- -1
- -1
- -1
- -1
- -1
- -1
- -1
- -1
- -1
- -1
- -1
- -1
- -1
- -1
- -1
- -1
- -1
- -1
- -1
- -1
- -1
- -1
- -1
- -1
- -1
- -1
- -1
- -1
- -1
- -1
- -1
- -1
- -1
- -1
- -1
- -1
- -1
- -1
- -1
- -1
- -1
- -1
- -1
- -1
- -1
- -1
- -1
- 0
- 100
- 0
- -1
- -1
- -1
- -1
- -1
- -1
- -1
- -1
- -1
- -1
- -1
- -1
- -1
- -1
- -1
- -1
- -1
- -1
- -1
- -1
- -1
- -1
- -1
- -1
- -1
- -1
- -1
- -1
- -1
- -1
- -1
- -1
- -1
- -1
- -1
- -1
- -1
- -1
- -1
- -1
- -1
- -1
- -1
- -1
- -1
- -1
- -1
- -1
- -1
- -1
- -1
- -1
- -1
- -1
- -1
- -1
- -1
- -1
- -1
- -1
- -1
- -1
- -1
- '...'
---
```

ahcorde, 2025-05-05:
Thank you for testing!

closed 2025-05-05 by ahcorde