ros2/kilted_tutorial_party #135: Dummy Sensors executables

tfoote, 2025-05-01:
## Setup
- Rmw: Cyclone
- BuildType: Binary
- Os: Windows
- Chip: Amd64

## Links
- [Dummy Sensors](https://github.com/ros2/demos/blob/kilted/dummy_robot/dummy_sensors/README.md#run)

## Checks
- [ ] **The executables in the README are present and work as expected.**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

ShahazadAbdulla, 2025-05-04:
Successfully completed all checks for Issue #135 on Windows 11 / amd64 / binary / CycloneDDS.

1.  **`dummy_laser` Check:**
    *   Ran `ros2 run dummy_sensors dummy_laser`. The expected initial INFO messages about scan parameters were printed.
    *   Ran `ros2 topic echo /scan` in a separate terminal. `sensor_msgs/msg/LaserScan` messages were continuously published and echoed correctly.

2.  **`dummy_joint_states` Check:**
    *   Ran `ros2 run dummy_sensors dummy_joint_states`.
    *   Ran `ros2 topic echo /joint_states` in a separate terminal. `sensor_msgs/msg/JointState` messages were continuously published with changing position values and echoed correctly.

All executables started and published data as expected.

ahcorde, 2025-05-05:
Thank you for testing

closed 2025-05-05 by ahcorde