ros2/kilted_tutorial_party #137: Dummy Sensors executables

tfoote, 2025-05-01:
## Setup
- Rmw: Fastdds
- BuildType: Binary
- Os: Windows
- Chip: Amd64

## Links
- [Dummy Sensors](https://github.com/ros2/demos/blob/kilted/dummy_robot/dummy_sensors/README.md#run)

## Checks
- [ ] **The executables in the README are present and work as expected.**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

pradyum, 2025-05-02:
**The dummy sensor executables are working fine**

Node
```
C:\pixi_ws>ros2 run dummy_sensors dummy_laser
[INFO] [1746229945.339054900] [dummy_laser]: angle inc: 0.004363
[INFO] [1746229945.342183800] [dummy_laser]: scan size: 1081
[INFO] [1746229945.343223700] [dummy_laser]: scan time increment:       0.000000
```

Output
```
C:\pixi_ws>ros2 topic echo /scan --once
header:
  stamp:
    sec: 1746230083
    nanosec: 315155100
  frame_id: single_rrbot_hokuyo_link
angle_min: -2.356194496154785
angle_max: 2.356194496154785
angle_increment: 0.004363323096185923
time_increment: 0.0
scan_time: 0.03999999910593033
range_min: 0.0
range_max: 10.0
ranges:
- 0.049535639584064484
- 0.049535639584064484
- 0.049535639584064484
- 0.049535639584064484
- 0.049535639584064484
- 0.049535639584064484
- 0.049535639584064484
- 0.049535639584064484
- 0.049535639584064484
- 0.049535639584064484
- 0.049535639584064484
- 0.049535639584064484
- 0.049535639584064484
- 0.049535639584064484
- 0.049535639584064484
- 0.049535639584064484
- 0.049535639584064484
- 0.049535639584064484
- 0.049535639584064484
- 0.049535639584064484
- 0.049535639584064484
- 0.049535639584064484
- 0.049535639584064484
- 0.049535639584064484
- 0.049535639584064484
- 0.049535639584064484
- 0.049535639584064484
- 0.049535639584064484
- 0.049535639584064484
- 0.049535639584064484
- 0.049535639584064484
- 0.049535639584064484
- 0.049535639584064484
- 0.049535639584064484
- 0.049535639584064484
- 0.049535639584064484
- 0.049535639584064484
- 0.049535639584064484
- 0.049535639584064484
- 0.049535639584064484
- 0.049535639584064484
- 0.049535639584064484
- 0.049535639584064484
- 0.049535639584064484
- 0.049535639584064484
- 0.049535639584064484
- 0.049535639584064484
- 0.049535639584064484
- 0.049535639584064484
- 0.049535639584064484
- 0.049535639584064484
- 0.049535639584064484
- 0.049535639584064484
- 0.049535639584064484
- 0.049535639584064484
- 0.049535639584064484
- 0.049535639584064484
- 0.049535639584064484
- 0.049535639584064484
- 0.049535639584064484
- 0.049535639584064484
- 0.049535639584064484
- 0.049535639584064484
- 0.049535639584064484
- 0.049535639584064484
- 0.049535639584064484
- 0.049535639584064484
- 0.049535639584064484
- 0.049535639584064484
- 0.049535639584064484
- 0.049535639584064484
- 0.049535639584064484
- 0.049535639584064484
- 0.049535639584064484
- 0.049535639584064484
- 0.049535639584064484
- 0.049535639584064484
- 0.049535639584064484
- 0.049535639584064484
- 0.049535639584064484
- 0.049535639584064484
- 0.049535639584064484
- 0.049535639584064484
- 0.049535639584064484
- 0.049535639584064484
- 0.049535639584064484
- 0.049535639584064484
- 0.049535639584064484
- 0.049535639584064484
- 0.049535639584064484
- 0.049535639584064484
- 0.049535639584064484
- 0.049535639584064484
- 0.049535639584064484
- 0.049535639584064484
- 0.049535639584064484
- 0.049535639584064484
- 0.049535639584064484
- 0.049535639584064484
- 0.049535639584064484
- 0.049535639584064484
- 0.049535639584064484
- 0.049535639584064484
- 0.049535639584064484
- 0.049535639584064484
- 0.049535639584064484
- 0.049535639584064484
- 0.049535639584064484
- 0.049535639584064484
- 0.049535639584064484
- 0.049535639584064484
- 0.049535639584064484
- 0.049535639584064484
- 0.049535639584064484
- 0.049535639584064484
- 0.049535639584064484
- 0.049535639584064484
- 0.049535639584064484
- 0.049535639584064484
- 0.049535639584064484
- 0.049535639584064484
- 0.049535639584064484
- 0.049535639584064484
- 0.049535639584064484
- 0.049535639584064484
- 0.049535639584064484
- 0.049535639584064484
- 0.049535639584064484
- '...'
intensities: []
---
```

---

Node
```
C:\pixi_ws>ros2 run dummy_sensors dummy_joint_states
```

Output
```
C:\pixi_ws>ros2 topic list
/joint_states
/parameter_events
/rosout

(pixi_ros2_rolling) C:\pixi_ws>ros2 topic echo /joint_states --once
header:
  stamp:
    sec: 1746230197
    nanosec: 198740000
  frame_id: ''
name:
- single_rrbot_joint1
- single_rrbot_joint2
position:
- -0.09367550009022263
- -0.09367550009022263
velocity: []
effort: []
---
```

ahcorde, 2025-05-05:
Thank you for testing!