ros2/kilted_tutorial_party #141: Executables in `dummy_robot_bringup`

tfoote, 2025-05-01:
## Setup
- Rmw: Cyclone
- BuildType: Binary
- Os: Windows
- Chip: Amd64

## Checks
- [ ] **Check `dummy_robot_bringup_launch.py`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 launch dummy_robot_bringup dummy_robot_bringup_launch.py
       ```
  
    
  </details>
- [ ] **Check `dummy_robot_bringup_launch.xml`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 launch dummy_robot_bringup dummy_robot_bringup_launch.xml
       ```
  
    
  </details>
- [ ] **Check `dummy_robot_bringup_launch.yaml`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 launch dummy_robot_bringup dummy_robot_bringup_launch.yaml
       ```
  
    
  </details>

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

ShahazadAbdulla, 2025-05-04:
I'll test this on Windows 11 / amd64 / binary / CycloneDDS

ShahazadAbdulla, 2025-05-04:
Successfully completed checks for Issue #141 on Windows 11 / amd64 / binary installation using CycloneDDS (`rmw_cyclonedds_cpp`).

- `ros2 launch dummy_robot_bringup dummy_robot_bringup_launch.py`: Launched successfully, started expected nodes (dummy_map_server, robot_state_publisher, etc.).
- `ros2 launch dummy_robot_bringup dummy_robot_bringup_launch.xml`: Launched successfully, started expected nodes.
- `ros2 launch dummy_robot_bringup dummy_robot_bringup_launch.yaml`: Launched successfully, started expected nodes.

(Note: A non-critical warning `UserWarning: Cannot create a symlink to latest log directory...` appeared during launch on Windows, but did not affect functionality.)

All checks passed.

closed 2025-05-05 by ahcorde