ros2/kilted_tutorial_party #143: Executables in `dummy_robot_bringup`

tfoote, 2025-05-01:
## Setup
- Rmw: Fastdds
- BuildType: Binary
- Os: Windows
- Chip: Amd64

## Checks
- [ ] **Check `dummy_robot_bringup_launch.py`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 launch dummy_robot_bringup dummy_robot_bringup_launch.py
       ```
  
    
  </details>
- [ ] **Check `dummy_robot_bringup_launch.xml`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 launch dummy_robot_bringup dummy_robot_bringup_launch.xml
       ```
  
    
  </details>
- [ ] **Check `dummy_robot_bringup_launch.yaml`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 launch dummy_robot_bringup dummy_robot_bringup_launch.yaml
       ```
  
    
  </details>

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

pradyum, 2025-05-02:
**The launch files and executables are working fine**

Check `dummy_robot_bringup_launch.py`

```
C:\pixi_ws>ros2 launch dummy_robot_bringup dummy_robot_bringup_launch.py
C:\pixi_ws\ros2-windows\Lib\site-packages\launch\logging\__init__.py:149: UserWarning: Cannot create a symlink to latest log directory: [WinError 1314] A required privilege is not held by the client: 'C:\\Users\\pc1\\.ros\\log\\2025-05-02-16-47-46-672019-DESKTOP-8BMD1U3-2836' -> 'C:\\Users\\pc1\\.ros\\log\\latest'

  warnings.warn(
[INFO] [launch]: All log files can be found below C:\Users\pc1\.ros\log\2025-05-02-16-47-46-672019-DESKTOP-8BMD1U3-2836
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [dummy_map_server.EXE-1]: process started with pid [3712]
[INFO] [robot_state_publisher.EXE-2]: process started with pid [6476]
[INFO] [dummy_joint_states.EXE-3]: process started with pid [6512]
[INFO] [dummy_laser.EXE-4]: process started with pid [6648]
[robot_state_publisher.EXE-2] [INFO] [1746229667.047706300] [robot_state_publisher]: Robot initialized
[dummy_laser.EXE-4] [INFO] [1746229667.065179300] [dummy_laser]: angle inc:     0.004363
[dummy_laser.EXE-4] [INFO] [1746229667.065279100] [dummy_laser]: scan size:     1081
[dummy_laser.EXE-4] [INFO] [1746229667.065288300] [dummy_laser]: scan time increment:   0.000000
```

Check `dummy_robot_bringup_launch.xml`

```
C:\pixi_ws>ros2 launch dummy_robot_bringup dummy_robot_bringup_launch.xml
C:\pixi_ws\ros2-windows\Lib\site-packages\launch\logging\__init__.py:149: UserWarning: Cannot create a symlink to latest log directory: [WinError 1314] A required privilege is not held by the client: 'C:\\Users\\pc1\\.ros\\log\\2025-05-02-16-49-14-838429-DESKTOP-8BMD1U3-6660' -> 'C:\\Users\\pc1\\.ros\\log\\latest'

  warnings.warn(
[INFO] [launch]: All log files can be found below C:\Users\pc1\.ros\log\2025-05-02-16-49-14-838429-DESKTOP-8BMD1U3-6660
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [dummy_map_server.EXE-1]: process started with pid [6296]
[INFO] [robot_state_publisher.EXE-2]: process started with pid [1996]
[INFO] [dummy_joint_states.EXE-3]: process started with pid [6008]
[INFO] [dummy_laser.EXE-4]: process started with pid [6420]
[dummy_laser.EXE-4] [INFO] [1746229756.668024400] [dummy_laser]: angle inc:     0.004363
[dummy_laser.EXE-4] [INFO] [1746229756.668498500] [dummy_laser]: scan size:     1081
[dummy_laser.EXE-4] [INFO] [1746229756.668510900] [dummy_laser]: scan time increment:   0.000000
[robot_state_publisher.EXE-2] [INFO] [1746229756.782374500] [robot_state_publisher]: Robot initialized
```

Check `dummy_robot_bringup_launch.yaml`

```
C:\pixi_ws>ros2 launch dummy_robot_bringup dummy_robot_bringup_launch.yaml
C:\pixi_ws\ros2-windows\Lib\site-packages\launch\logging\__init__.py:149: UserWarning: Cannot create a symlink to latest log directory: [WinError 1314] A required privilege is not held by the client: 'C:\\Users\\pc1\\.ros\\log\\2025-05-02-16-50-29-210273-DESKTOP-8BMD1U3-4676' -> 'C:\\Users\\pc1\\.ros\\log\\latest'

  warnings.warn(
[INFO] [launch]: All log files can be found below C:\Users\pc1\.ros\log\2025-05-02-16-50-29-210273-DESKTOP-8BMD1U3-4676
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [dummy_map_server.EXE-1]: process started with pid [3172]
[INFO] [robot_state_publisher.EXE-2]: process started with pid [2580]
[INFO] [dummy_joint_states.EXE-3]: process started with pid [7744]
[INFO] [dummy_laser.EXE-4]: process started with pid [4456]
[dummy_laser.EXE-4] [INFO] [1746229829.854677400] [dummy_laser]: angle inc:     0.004363
[dummy_laser.EXE-4] [INFO] [1746229829.855184800] [dummy_laser]: scan size:     1081
[dummy_laser.EXE-4] [INFO] [1746229829.855213000] [dummy_laser]: scan time increment:   0.000000
[robot_state_publisher.EXE-2] [INFO] [1746229829.942270200] [robot_state_publisher]: Robot initialized
```

For all the check output of `ros2 node list` is 

```
C:\pixi_ws>ros2 node list
/dummy_joint_states
/dummy_laser
/dummy_map_server
/robot_state_publisher
```

ahcorde, 2025-05-05:
Thank you for testing!

closed 2025-05-05 by ahcorde