ros2/kilted_tutorial_party #147: Executables in `examples_rclcpp_minimal_action_client`

tfoote, 2025-05-01:
## Setup
- Rmw: Cyclone
- BuildType: Debian
- Os: Noble
- Chip: Amd64

## Checks
- [ ] **Check `action_client_not_composable`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run examples_rclcpp_minimal_action_client action_client_not_composable
       ```
  
    
  </details>
- [ ] **Check `action_client_not_composable_with_cancel`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run examples_rclcpp_minimal_action_client action_client_not_composable_with_cancel
       ```
  
    
  </details>
- [ ] **Check `action_client_not_composable_with_feedback`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run examples_rclcpp_minimal_action_client action_client_not_composable_with_feedback
       ```
  
    
  </details>
- [ ] **Check `action_client_member_functions`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run examples_rclcpp_minimal_action_client action_client_member_functions
       ```
  
    
  </details>

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

suchetanrs, 2025-05-03:
It works for me.

The script to run the test can be found here https://github.com/suchetanrs/kilted_tutorial_party/tree/main/scripts

```
root@cluttered:~# uname -m
x86_64
root@cluttered:~# source /opt/ros/kilted/setup.bash
root@cluttered:~# export RMW_IMPLEMENTATION=rmw_cyclonedds_cpp
root@cluttered:~# ros2 doctor --report | grep middleware
middleware name    : rmw_cyclonedds_cpp
root@cluttered:~# ./scripts/examples_rclcpp_minimal_action_client.sh 
====================================================running the server
=========================================================
=========================================================
=========================================================
====================================================running the action_client_not_composable
=========================================================
=========================================================
=========================================================
[INFO] [1746283329.072545406] [minimal_action_client]: Sending goal
[INFO] [1746283329.072637373] [minimal_action_server]: Received goal request with order 10
[INFO] [1746283329.072749143] [minimal_action_server]: Executing goal
[INFO] [1746283329.072780906] [minimal_action_client]: Waiting for result
[INFO] [1746283329.072802518] [minimal_action_server]: Publish Feedback
[INFO] [1746283330.072886744] [minimal_action_server]: Publish Feedback
[INFO] [1746283331.072884763] [minimal_action_server]: Publish Feedback
[INFO] [1746283332.072884012] [minimal_action_server]: Publish Feedback
[INFO] [1746283333.072911602] [minimal_action_server]: Publish Feedback
[INFO] [1746283334.072913395] [minimal_action_server]: Publish Feedback
[INFO] [1746283335.072918237] [minimal_action_server]: Publish Feedback
[INFO] [1746283336.072916348] [minimal_action_server]: Publish Feedback
[INFO] [1746283337.072889168] [minimal_action_server]: Publish Feedback
[INFO] [1746283338.073086983] [minimal_action_server]: Goal Succeeded
[INFO] [1746283338.073387681] [minimal_action_client]: result received
[INFO] [1746283338.073411538] [minimal_action_client]: 0
[INFO] [1746283338.073416975] [minimal_action_client]: 1
[INFO] [1746283338.073426443] [minimal_action_client]: 1
[INFO] [1746283338.073430612] [minimal_action_client]: 2
[INFO] [1746283338.073433400] [minimal_action_client]: 3
[INFO] [1746283338.073438642] [minimal_action_client]: 5
[INFO] [1746283338.073442751] [minimal_action_client]: 8
[INFO] [1746283338.073446177] [minimal_action_client]: 13
[INFO] [1746283338.073450026] [minimal_action_client]: 21
[INFO] [1746283338.073453583] [minimal_action_client]: 34
[INFO] [1746283338.073456850] [minimal_action_client]: 55
====================================================running the action_client_not_composable_with_cancel
=========================================================
=========================================================
=========================================================
[INFO] [1746283338.443737623] [minimal_action_client]: Sending goal
[INFO] [1746283338.443846101] [minimal_action_server]: Received goal request with order 10
[INFO] [1746283338.443959437] [minimal_action_server]: Executing goal
[INFO] [1746283338.444000021] [minimal_action_server]: Publish Feedback
[INFO] [1746283339.444089553] [minimal_action_server]: Publish Feedback
[INFO] [1746283340.444134695] [minimal_action_server]: Publish Feedback
[INFO] [1746283341.444179307] [minimal_action_client]: canceling goal
[INFO] [1746283341.444184894] [minimal_action_server]: Publish Feedback
[INFO] [1746283341.444412464] [minimal_action_server]: Received request to cancel goal
[INFO] [1746283341.444522591] [minimal_action_client]: goal is being canceled
[INFO] [1746283341.444538692] [minimal_action_client]: Waiting for result
[INFO] [1746283342.444283084] [minimal_action_server]: Goal Canceled
[ERROR] [1746283342.444582861] [minimal_action_client]: Goal was canceled
[ros2run]: Process exited with failure 1
====================================================running the action_client_not_composable_with_feedback
=========================================================
=========================================================
=========================================================
[INFO] [1746283342.810215546] [minimal_action_client]: Sending goal
[INFO] [1746283342.810359856] [minimal_action_server]: Received goal request with order 10
[... 49 lines cut ...]
[INFO] [1746283356.679068872] [minimal_action_server]: Publish Feedback
[INFO] [1746283356.679188677] [minimal_action_client]: Next number in sequence received: 8
[INFO] [1746283357.679032037] [minimal_action_server]: Publish Feedback
[INFO] [1746283357.679156391] [minimal_action_client]: Next number in sequence received: 13
[INFO] [1746283358.679107660] [minimal_action_server]: Publish Feedback
[INFO] [1746283358.679248023] [minimal_action_client]: Next number in sequence received: 21
[INFO] [1746283359.679035825] [minimal_action_server]: Publish Feedback
[INFO] [1746283359.679130219] [minimal_action_client]: Next number in sequence received: 34
[INFO] [1746283360.679080990] [minimal_action_server]: Publish Feedback
[INFO] [1746283360.679163657] [minimal_action_client]: Next number in sequence received: 55
[INFO] [1746283361.679148788] [minimal_action_server]: Goal Succeeded
[INFO] [1746283361.679231037] [minimal_action_client]: Result received
[INFO] [1746283361.679253014] [minimal_action_client]: 0
[... cut ...]
[INFO] [1746283361.679263143] [minimal_action_client]: 1
[INFO] [1746283361.679267203] [minimal_action_client]: 2
[INFO] [1746283361.679270873] [minimal_action_client]: 3
[INFO] [1746283361.679275037] [minimal_action_client]: 5
[INFO] [1746283361.679278280] [minimal_action_client]: 8
[INFO] [1746283361.679287527] [minimal_action_client]: 13
[INFO] [1746283361.679296876] [minimal_action_client]: 21
[INFO] [1746283361.679300018] [minimal_action_client]: 34
[INFO] [1746283361.679303117] [minimal_action_client]: 55
root@cluttered:~# 
```

ahcorde, 2025-05-05:
Thank you for testing