ros2/kilted_tutorial_party #149: Executables in `examples_rclcpp_minimal_action_client`

tfoote, 2025-05-01:
## Setup
- Rmw: Fastdds
- BuildType: Debian
- Os: Noble
- Chip: Amd64

## Checks
- [ ] **Check `action_client_not_composable`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run examples_rclcpp_minimal_action_client action_client_not_composable
       ```
  
    
  </details>
- [ ] **Check `action_client_not_composable_with_cancel`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run examples_rclcpp_minimal_action_client action_client_not_composable_with_cancel
       ```
  
    
  </details>
- [ ] **Check `action_client_not_composable_with_feedback`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run examples_rclcpp_minimal_action_client action_client_not_composable_with_feedback
       ```
  
    
  </details>
- [ ] **Check `action_client_member_functions`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run examples_rclcpp_minimal_action_client action_client_member_functions
       ```
  
    
  </details>

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

suchetanrs, 2025-05-03:
- [x] Check action_client_not_composable

Script to run can be found at https://github.com/suchetanrs/kilted_tutorial_party/tree/main/scripts

```
root@cluttered:~# source /opt/ros/kilted/setup.bash
root@cluttered:~# uname -m
x86_64
root@cluttered:~# ros2 doctor --report | grep middleware
middleware name    : rmw_fastrtps_cpp
root@cluttered:~# ./scripts/examples_rclcpp_minimal_action_client.sh 
====================================================running the server
=========================================================
=========================================================
=========================================================
====================================================running the action_client_not_composable
=========================================================
=========================================================
=========================================================
[INFO] [1746283093.701866020] [minimal_action_client]: Sending goal
[INFO] [1746283093.886236281] [minimal_action_server]: Received goal request with order 10
[INFO] [1746283093.886585693] [minimal_action_server]: Executing goal
[INFO] [1746283093.886688041] [minimal_action_server]: Publish Feedback
[INFO] [1746283093.886759852] [minimal_action_client]: Waiting for result
[INFO] [1746283094.886880419] [minimal_action_server]: Publish Feedback
[INFO] [1746283095.886817322] [minimal_action_server]: Publish Feedback
[INFO] [1746283096.886750336] [minimal_action_server]: Publish Feedback
[INFO] [1746283097.886851930] [minimal_action_server]: Publish Feedback
[INFO] [1746283098.886796703] [minimal_action_server]: Publish Feedback
[INFO] [1746283099.886875421] [minimal_action_server]: Publish Feedback
[INFO] [1746283100.886867070] [minimal_action_server]: Publish Feedback
[INFO] [1746283101.886778440] [minimal_action_server]: Publish Feedback
[INFO] [1746283102.886919844] [minimal_action_server]: Goal Succeeded
[INFO] [1746283102.887164846] [minimal_action_client]: result received
[INFO] [1746283102.887190785] [minimal_action_client]: 0
[INFO] [1746283102.887194474] [minimal_action_client]: 1
[INFO] [1746283102.887222060] [minimal_action_client]: 1
[INFO] [1746283102.887224968] [minimal_action_client]: 2
[INFO] [1746283102.887227735] [minimal_action_client]: 3
[INFO] [1746283102.887231041] [minimal_action_client]: 5
[INFO] [1746283102.887233684] [minimal_action_client]: 8
[INFO] [1746283102.887236315] [minimal_action_client]: 13
[INFO] [1746283102.887239073] [minimal_action_client]: 21
[INFO] [1746283102.887241804] [minimal_action_client]: 34
[INFO] [1746283102.887244644] [minimal_action_client]: 55
====================================================running the action_client_not_composable_with_cancel
=========================================================
=========================================================
=========================================================
[INFO] [1746283103.359104550] [minimal_action_client]: Sending goal
[INFO] [1746283103.359470264] [minimal_action_server]: Received goal request with order 10
[INFO] [1746283103.359611199] [minimal_action_server]: Executing goal
[INFO] [1746283103.359690252] [minimal_action_server]: Publish Feedback
[INFO] [1746283104.359764352] [minimal_action_server]: Publish Feedback
[INFO] [1746283105.359894891] [minimal_action_server]: Publish Feedback
[INFO] [1746283106.359737193] [minimal_action_server]: Publish Feedback
[INFO] [1746283106.359814684] [minimal_action_client]: canceling goal
[INFO] [1746283106.360024729] [minimal_action_server]: Received request to cancel goal
[INFO] [1746283106.360293877] [minimal_action_client]: goal is being canceled
[INFO] [1746283106.360320001] [minimal_action_client]: Waiting for result
[INFO] [1746283107.359920903] [minimal_action_server]: Goal Canceled
[ERROR] [1746283107.360274417] [minimal_action_client]: Goal was canceled
[ros2run]: Process exited with failure 1
====================================================running the action_client_not_composable_with_feedback
=========================================================
=========================================================
=========================================================
[INFO] [1746283107.744879308] [minimal_action_client]: Sending goal
[INFO] [1746283107.745038930] [minimal_action_server]: Received goal request with order 10
[INFO] [1746283107.745125801] [minimal_action_server]: Executing goal
[... 47 lines cut ...]
[INFO] [1746283121.532031086] [minimal_action_server]: Publish Feedback
[INFO] [1746283121.532157493] [minimal_action_client]: Next number in sequence received: 8
[INFO] [1746283122.532036773] [minimal_action_server]: Publish Feedback
[INFO] [1746283122.532101329] [minimal_action_client]: Next number in sequence received: 13
[INFO] [1746283123.531957915] [minimal_action_server]: Publish Feedback
[INFO] [1746283123.532001672] [minimal_action_client]: Next number in sequence received: 21
[INFO] [1746283124.532049737] [minimal_action_server]: Publish Feedback
[INFO] [1746283124.532151223] [minimal_action_client]: Next number in sequence received: 34
[INFO] [1746283125.532049364] [minimal_action_server]: Publish Feedback
[INFO] [1746283125.532165303] [minimal_action_client]: Next number in sequence received: 55
[INFO] [1746283126.532023863] [minimal_action_server]: Goal Succeeded
[INFO] [1746283126.532231151] [minimal_action_client]: Result received
[INFO] [1746283126.532255650] [minimal_action_client]: 0
[... cut ...]
[INFO] [1746283126.532278234] [minimal_action_client]: 1
[INFO] [1746283126.532286953] [minimal_action_client]: 2
[INFO] [1746283126.532289932] [minimal_action_client]: 3
[INFO] [1746283126.532292926] [minimal_action_client]: 5
[INFO] [1746283126.532295784] [minimal_action_client]: 8
[INFO] [1746283126.532298322] [minimal_action_client]: 13
[INFO] [1746283126.532301077] [minimal_action_client]: 21
[INFO] [1746283126.532303865] [minimal_action_client]: 34
[INFO] [1746283126.532306921] [minimal_action_client]: 55
root@cluttered:~# 
```

ahcorde, 2025-05-05:
Thank you for testing

closed 2025-05-05 by ahcorde