ros2/kilted_tutorial_party #150: Executables in `examples_rclcpp_minimal_action_client`

tfoote, 2025-05-01:
## Setup
- Rmw: Zenoh
- BuildType: Debian
- Os: Noble
- Chip: Amd64

## Checks
- [ ] **Check `action_client_not_composable`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run examples_rclcpp_minimal_action_client action_client_not_composable
       ```
  
    
  </details>
- [ ] **Check `action_client_not_composable_with_cancel`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run examples_rclcpp_minimal_action_client action_client_not_composable_with_cancel
       ```
  
    
  </details>
- [ ] **Check `action_client_not_composable_with_feedback`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run examples_rclcpp_minimal_action_client action_client_not_composable_with_feedback
       ```
  
    
  </details>
- [ ] **Check `action_client_member_functions`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run examples_rclcpp_minimal_action_client action_client_member_functions
       ```
  
    
  </details>

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

jncfa, 2025-05-02:
Tested with Ubuntu Docker Container for amd64(under Rosetta 2) (ghcr.io/sloretz/ros-testing:kilted-desktop)

Also tested `examples_rclcpp_minimal_action_server` to be able to test these executables:

[image]

[image]

ahcorde, 2025-05-02:
Thank you for testing

closed 2025-05-02 by ahcorde