ros2/kilted_tutorial_party #151: Executables in `examples_rclcpp_minimal_action_server`

tfoote, 2025-05-01:
## Setup
- Rmw: Cyclone
- BuildType: Binary
- Os: Windows
- Chip: Amd64

## Checks
- [ ] **Check `action_server_not_composable`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run examples_rclcpp_minimal_action_server action_server_not_composable
       ```
  
    
  </details>
- [ ] **Check `action_server_member_functions`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run examples_rclcpp_minimal_action_server action_server_member_functions
       ```
  
    
  </details>

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

ShahazadAbdulla, 2025-05-04:
I'll test this on Windows 11 / amd64 / binary / CycloneDDS

ShahazadAbdulla, 2025-05-04:
Uh oh! Didn't saw the self assigned of @mjcarroll there I was testing out a lot sorry.

anyway here are my findings if it helps you or the community, Thank you

Testing Issue #151 on Windows 11 / amd64 / binary / CycloneDDS (`rmw_cyclonedds_cpp`).

**Results:**
- `ros2 run examples_rclcpp_minimal_action_server action_server_not_composable`: Executable started successfully without crashing.
- `ros2 run examples_rclcpp_minimal_action_server action_server_member_functions`: Executable started successfully without crashing.

**However, attempting to interact with the servers using `ros2 action send_goal` FAILED:**
- Ran `ros2 action send_goal /fibonacci example_interfaces/action/Fibonacci "{order: 5}"` in a separate terminal while either server was running.
- The client printed `Waiting for an action server to become available...` followed by errors:
```2025-05-04 16:56:47.370 [RTPS_READER_HISTORY Error] Change payload size of '32' bytes is larger than the history payload size of '19' bytes and cannot be resized. -> Function eprosima::fastdds::rtps::ReaderHistory::can_change_be_added_nts
2025-05-04 16:56:47.373 [RTPS_READER_HISTORY Error] Change payload size of '32' bytes is larger than the history payload size of '19' bytes and cannot be resized. -> Function eprosima::fastdds::rtps::ReaderHistory::can_change_be_added_nts
```
- The action servers running in the other terminal did not show any indication of receiving the goal.

**Conclusion:** While the action server executables launch, there appears to be an underlying RMW/DDS issue preventing the `ros2 action send_goal` CLI tool from successfully sending a goal to them on this Windows/CycloneDDS setup, indicated by the RTPS payload size errors. The basic executability check could be considered passed, but the action functionality itself failed testing via the standard CLI tool.

mjcarroll, 2025-05-09:
> Uh oh! Didn't saw the self assigned of @mjcarroll there I was testing out a lot and didn't noticed you, sorry.

I don't recall assigning myself here, thanks for taking it!

closed 2025-05-09 by mjcarroll