ros2/kilted_tutorial_party #154: Executables in `examples_rclcpp_minimal_action_server`

tfoote, 2025-05-01:
## Setup
- Rmw: Fastdds
- BuildType: Debian
- Os: Noble
- Chip: Amd64

## Checks
- [ ] **Check `action_server_not_composable`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run examples_rclcpp_minimal_action_server action_server_not_composable
       ```
  
    
  </details>
- [ ] **Check `action_server_member_functions`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run examples_rclcpp_minimal_action_server action_server_member_functions
       ```
  
    
  </details>

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

suchetanrs, 2025-05-03:
Both of them work for me

[image]

Noel215, 2025-05-03:
Both checks seem fine.

---
Set up:

```
noel@laptop:~$ lsb_release -a
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 24.04.2 LTS
Release:	24.04
Codename:	noble
noel@laptop:~$ dpkg --print-architecture
amd64
noel@laptop:~$ printenv AMENT_PREFIX_PATH 
/opt/ros/kilted
noel@laptop:~$ printenv RMW_IMPLEMENTATION
noel@laptop:~$ 
```
---
**Check `action_server_not_composable`**

Server:
```
noel@laptop:~$ ros2 run examples_rclcpp_minimal_action_server action_server_not_composable
[INFO] [1746312264.683615693] [server]: Got goal request with order 10
[INFO] [1746312264.683924299] [server]: Executing goal
[INFO] [1746312264.683994050] [server]: Publish Feedback
[INFO] [1746312265.684429101] [server]: Publish Feedback
[INFO] [1746312266.684245773] [server]: Publish Feedback
[INFO] [1746312267.684212754] [server]: Publish Feedback
[INFO] [1746312268.684281381] [server]: Publish Feedback
[INFO] [1746312269.684211281] [server]: Publish Feedback
[INFO] [1746312270.684186451] [server]: Publish Feedback
[INFO] [1746312271.684225955] [server]: Publish Feedback
[INFO] [1746312272.684224833] [server]: Publish Feedback
[INFO] [1746312273.684541171] [server]: Goal Succeeded
```

Client:
```
noel@laptop:~$ ros2 run examples_rclcpp_minimal_action_client action_client_not_composable
[INFO] [1746312264.683304294] [minimal_action_client]: Sending goal
[INFO] [1746312264.683990683] [minimal_action_client]: Waiting for result
[INFO] [1746312273.684920812] [minimal_action_client]: result received
[INFO] [1746312273.685008754] [minimal_action_client]: 0
[INFO] [1746312273.685037766] [minimal_action_client]: 1
[INFO] [1746312273.685060101] [minimal_action_client]: 1
[INFO] [1746312273.685079639] [minimal_action_client]: 2
[INFO] [1746312273.685100086] [minimal_action_client]: 3
[INFO] [1746312273.685148150] [minimal_action_client]: 5
[INFO] [1746312273.685170285] [minimal_action_client]: 8
[INFO] [1746312273.685186780] [minimal_action_client]: 13
[INFO] [1746312273.685203965] [minimal_action_client]: 21
[INFO] [1746312273.685220299] [minimal_action_client]: 34
[INFO] [1746312273.685240878] [minimal_action_client]: 55
```
---
**Check `action_server_member_functions`**

Server:
```
noel@laptop:~$ ros2 run examples_rclcpp_minimal_action_server action_server_member_functions 
[INFO] [1746312463.040844494] [minimal_action_server]: Received goal request with order 10
[INFO] [1746312463.041055346] [minimal_action_server]: Executing goal
[INFO] [1746312463.041153282] [minimal_action_server]: Publish Feedback
[INFO] [1746312464.041372698] [minimal_action_server]: Publish Feedback
[INFO] [1746312465.041703854] [minimal_action_server]: Publish Feedback
[INFO] [1746312466.041374871] [minimal_action_server]: Publish Feedback
[INFO] [1746312467.041491445] [minimal_action_server]: Publish Feedback
[INFO] [1746312468.041264028] [minimal_action_server]: Publish Feedback
[INFO] [1746312469.041791689] [minimal_action_server]: Publish Feedback
[INFO] [1746312470.041726232] [minimal_action_server]: Publish Feedback
[INFO] [1746312471.041266299] [minimal_action_server]: Publish Feedback
[INFO] [1746312472.041784440] [minimal_action_server]: Goal Succeeded
```

Client:
```
noel@laptop:~$ ros2 run examples_rclcpp_minimal_action_client action_client_not_composable
[INFO] [1746312463.040560385] [minimal_action_client]: Sending goal
[INFO] [1746312463.041153069] [minimal_action_client]: Waiting for result
[INFO] [1746312472.042128135] [minimal_action_client]: result received
[INFO] [1746312472.042215849] [minimal_action_client]: 0
[INFO] [1746312472.042249129] [minimal_action_client]: 1
[INFO] [1746312472.042283797] [minimal_action_client]: 1
[INFO] [1746312472.042317328] [minimal_action_client]: 2
[INFO] [1746312472.042353039] [minimal_action_client]: 3
[INFO] [1746312472.042413587] [minimal_action_client]: 5
[INFO] [1746312472.042446263] [minimal_action_client]: 8
[INFO] [1746312472.042481366] [minimal_action_client]: 13
[INFO] [1746312472.042513618] [minimal_action_client]: 21
[INFO] [1746312472.042543447] [minimal_action_client]: 34
[INFO] [1746312472.042571994] [minimal_action_client]: 55
```

ahcorde, 2025-05-05:
Thank you for testing. Gracias 👍

closed 2025-05-05 by ahcorde