ros2/kilted_tutorial_party #156: Executables in `examples_rclcpp_minimal_client`

tfoote, 2025-05-01:
## Setup
- Rmw: Cyclone
- BuildType: Binary
- Os: Windows
- Chip: Amd64

## Checks
- [ ] **Check `client_main`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run examples_rclcpp_minimal_client client_main
       ```
  
    
  </details>

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

ShahazadAbdulla, 2025-05-04:
I'll test this on Windows 11 / amd64 / binary / CycloneDDS

ShahazadAbdulla, 2025-05-04:
Successfully completed checks for Issue #156 on Windows 11 / amd64 / binary installation using CycloneDDS (`rmw_cyclonedds_cpp`).

- Ran `ros2 run examples_rclcpp_minimal_client client_main` by itself. The client started correctly and waited for the `/add_two_ints` service as expected without crashing.
- Also tested by running `ros2 run demo_nodes_cpp add_two_ints_server` in a separate terminal first. The client then successfully called the service and printed the correct sum.

All checks passed.

ahcorde, 2025-05-05:
Thank you for testing

closed 2025-05-05 by ahcorde