ros2/kilted_tutorial_party #159: Executables in `examples_rclcpp_minimal_client`

tfoote, 2025-05-01:
## Setup
- Rmw: Fastdds
- BuildType: Debian
- Os: Noble
- Chip: Amd64

## Checks
- [ ] **Check `client_main`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run examples_rclcpp_minimal_client client_main
       ```
  
    
  </details>

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

ivrolan, 2025-05-02:
Works as expected using the Ubuntu Docker Container (from Ubuntu 22.04 host amd64). I ran the service from #174 besides to test if the request are received properly:

[image]

ahcorde, 2025-05-02:
Thank you for testing

closed 2025-05-02 by ahcorde