ros2/kilted_tutorial_party #162: Executables in `examples_rclcpp_minimal_composition`

tfoote, 2025-05-01:
## Setup
- Rmw: Cyclone
- BuildType: Debian
- Os: Noble
- Chip: Amd64

## Checks
- [ ] **Check `composition_composed`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run examples_rclcpp_minimal_composition composition_composed
       ```
  
    
  </details>
- [ ] **Check `composition_publisher`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run examples_rclcpp_minimal_composition composition_publisher
       ```
  
    
  </details>
- [ ] **Check `composition_subscriber`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run examples_rclcpp_minimal_composition composition_subscriber
       ```
  
    
  </details>

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

suchetanrs, 2025-05-02:
All of them work fine

```
root@cluttered:~# tmyx
bash: tmyx: command not found
root@cluttered:~# tmux
[exited]
root@cluttered:~# uname -m
x86_64
root@cluttered:~# source /opt/ros/kilted/setup.bash
root@cluttered:~# export RMW_IMPLEMENTATION=rmw_cyclonedds_cpp
root@cluttered:~# # User input in terminal 1
ros2 run examples_rclcpp_minimal_composition composition_composed
[INFO] [1746219579.894463680] [publisher_node]: Publisher: 'Hello, world! 0'
[INFO] [1746219579.894649726] [subscriber_node]: Subscriber: 'Hello, world! 0'
[INFO] [1746219580.394507601] [publisher_node]: Publisher: 'Hello, world! 1'
[INFO] [1746219580.394605846] [subscriber_node]: Subscriber: 'Hello, world! 1'
[INFO] [1746219580.894496983] [publisher_node]: Publisher: 'Hello, world! 2'
[INFO] [1746219580.894576677] [subscriber_node]: Subscriber: 'Hello, world! 2'
[INFO] [1746219581.394487054] [publisher_node]: Publisher: 'Hello, world! 3'
[INFO] [1746219581.394553314] [subscriber_node]: Subscriber: 'Hello, world! 3'
[INFO] [1746219581.894422034] [publisher_node]: Publisher: 'Hello, world! 4'
[INFO] [1746219581.894530489] [subscriber_node]: Subscriber: 'Hello, world! 4'
[INFO] [1746219582.394414383] [publisher_node]: Publisher: 'Hello, world! 5'
[INFO] [1746219582.394499419] [subscriber_node]: Subscriber: 'Hello, world! 5'
^C[INFO] [1746219582.460006179] [rclcpp]: signal_handler(signum=2)
[ros2run]: Received signal:  Interrupt
[INFO] [1746219582.460088007] [rclcpp]: signal_handler(signum=2)
```

[image]

ahcorde, 2025-05-02:
Thank you for testing

closed 2025-05-02 by ahcorde