ros2/kilted_tutorial_party #163: Executables in `examples_rclcpp_minimal_composition`

tfoote, 2025-05-01:
## Setup
- Rmw: Fastdds
- BuildType: Binary
- Os: Windows
- Chip: Amd64

## Checks
- [ ] **Check `composition_composed`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run examples_rclcpp_minimal_composition composition_composed
       ```
  
    
  </details>
- [ ] **Check `composition_publisher`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run examples_rclcpp_minimal_composition composition_publisher
       ```
  
    
  </details>
- [ ] **Check `composition_subscriber`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run examples_rclcpp_minimal_composition composition_subscriber
       ```
  
    
  </details>

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

pradyum, 2025-05-02:
**The executables are working fine**

Check composition_composed

```
C:\pixi_ws>ros2 run examples_rclcpp_minimal_composition composition_composed
[INFO] [1746228998.112129400] [publisher_node]: Publisher: 'Hello, world! 0'
[INFO] [1746228998.114791900] [subscriber_node]: Subscriber: 'Hello, world! 0'
[INFO] [1746228998.612295200] [publisher_node]: Publisher: 'Hello, world! 1'
[INFO] [1746228998.614952700] [subscriber_node]: Subscriber: 'Hello, world! 1'
[INFO] [1746228999.112018800] [publisher_node]: Publisher: 'Hello, world! 2'
[INFO] [1746228999.113015400] [subscriber_node]: Subscriber: 'Hello, world! 2'
[INFO] [1746228999.627417000] [publisher_node]: Publisher: 'Hello, world! 3'
[INFO] [1746228999.630475400] [subscriber_node]: Subscriber: 'Hello, world! 3'
[INFO] [1746229000.111800600] [publisher_node]: Publisher: 'Hello, world! 4'
[INFO] [1746229000.115150900] [subscriber_node]: Subscriber: 'Hello, world! 4'
[INFO] [1746229000.611577500] [publisher_node]: Publisher: 'Hello, world! 5'
[INFO] [1746229000.613907700] [subscriber_node]: Subscriber: 'Hello, world! 5'
[INFO] [1746229001.112371000] [publisher_node]: Publisher: 'Hello, world! 6'
[INFO] [1746229001.114390700] [subscriber_node]: Subscriber: 'Hello, world! 6'
[INFO] [1746229001.612337000] [publisher_node]: Publisher: 'Hello, world! 7'
[INFO] [1746229001.612634300] [subscriber_node]: Subscriber: 'Hello, world! 7'
```

Check composition_publisher

```
C:\pixi_ws>ros2 run examples_rclcpp_minimal_composition composition_publisher
[INFO] [1746229043.687589200] [publisher_node]: Publisher: 'Hello, world! 0'
[INFO] [1746229044.190956700] [publisher_node]: Publisher: 'Hello, world! 1'
[INFO] [1746229044.694641700] [publisher_node]: Publisher: 'Hello, world! 2'
[INFO] [1746229045.194799100] [publisher_node]: Publisher: 'Hello, world! 3'
[INFO] [1746229045.695661700] [publisher_node]: Publisher: 'Hello, world! 4'
[INFO] [1746229046.195569100] [publisher_node]: Publisher: 'Hello, world! 5'
[INFO] [1746229046.695214100] [publisher_node]: Publisher: 'Hello, world! 6'
[INFO] [1746229047.912893400] [publisher_node]: Publisher: 'Hello, world! 7'
[INFO] [1746229048.195438900] [publisher_node]: Publisher: 'Hello, world! 8'
[INFO] [1746229048.695657400] [publisher_node]: Publisher: 'Hello, world! 9'
[INFO] [1746229049.195300200] [publisher_node]: Publisher: 'Hello, world! 10'
```

Check composition_subscriber

```
C:\pixi_ws>ros2 run examples_rclcpp_minimal_composition composition_subscriber
[INFO] [1746229043.688849200] [subscriber_node]: Subscriber: 'Hello, world! 0'
[INFO] [1746229044.193063100] [subscriber_node]: Subscriber: 'Hello, world! 1'
[INFO] [1746229044.697094100] [subscriber_node]: Subscriber: 'Hello, world! 2'
[INFO] [1746229045.196917500] [subscriber_node]: Subscriber: 'Hello, world! 3'
[INFO] [1746229045.697448600] [subscriber_node]: Subscriber: 'Hello, world! 4'
[INFO] [1746229046.197132000] [subscriber_node]: Subscriber: 'Hello, world! 5'
[INFO] [1746229047.912840800] [subscriber_node]: Subscriber: 'Hello, world! 6'
[INFO] [1746229047.917092500] [subscriber_node]: Subscriber: 'Hello, world! 7'
[INFO] [1746229048.197192800] [subscriber_node]: Subscriber: 'Hello, world! 8'
[INFO] [1746229048.697373800] [subscriber_node]: Subscriber: 'Hello, world! 9'
[INFO] [1746229049.196357600] [subscriber_node]: Subscriber: 'Hello, world! 10'
```

ahcorde, 2025-05-05:
Thank you for testing

closed 2025-05-05 by ahcorde