ros2/kilted_tutorial_party #164: Executables in `examples_rclcpp_minimal_composition`

tfoote, 2025-05-01:
## Setup
- Rmw: Fastdds
- BuildType: Debian
- Os: Noble
- Chip: Amd64

## Checks
- [ ] **Check `composition_composed`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run examples_rclcpp_minimal_composition composition_composed
       ```
  
    
  </details>
- [ ] **Check `composition_publisher`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run examples_rclcpp_minimal_composition composition_publisher
       ```
  
    
  </details>
- [ ] **Check `composition_subscriber`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run examples_rclcpp_minimal_composition composition_subscriber
       ```
  
    
  </details>

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

suchetanrs, 2025-05-02:
```
root@cluttered:~# source /opt/ros/kilted/setup.bash
root@cluttered:~# ros2 doctor --report | grep middleware
middleware name    : rmw_fastrtps_cpp
root@cluttered:~# uname -m
x86_64
root@cluttered:~# # User input in terminal 1
ros2 run examples_rclcpp_minimal_composition composition_composed
[INFO] [1746219710.334657788] [publisher_node]: Publisher: 'Hello, world! 0'
[INFO] [1746219710.334896881] [subscriber_node]: Subscriber: 'Hello, world! 0'
[INFO] [1746219710.834649560] [publisher_node]: Publisher: 'Hello, world! 1'
[INFO] [1746219710.834804424] [subscriber_node]: Subscriber: 'Hello, world! 1'
^C[INFO] [1746219711.255065732] [rclcpp]: signal_handler(signum=2)
[ros2run]: Received signal:  Interrupt
[INFO] [1746219711.255232690] [rclcpp]: signal_handler(signum=2)
root@cluttered:~# 
```

[image]

ahcorde, 2025-05-02:
Thank you for testing

closed 2025-05-02 by ahcorde