ros2/kilted_tutorial_party #165: Executables in `examples_rclcpp_minimal_composition`

tfoote, 2025-05-01:
## Setup
- Rmw: Zenoh
- BuildType: Debian
- Os: Noble
- Chip: Amd64

## Checks
- [ ] **Check `composition_composed`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run examples_rclcpp_minimal_composition composition_composed
       ```
  
    
  </details>
- [ ] **Check `composition_publisher`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run examples_rclcpp_minimal_composition composition_publisher
       ```
  
    
  </details>
- [ ] **Check `composition_subscriber`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run examples_rclcpp_minimal_composition composition_subscriber
       ```
  
    
  </details>

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

jncfa, 2025-05-02:
Tested with Ubuntu Docker Container for amd64(under Rosetta 2) (ghcr.io/sloretz/ros-testing:kilted-desktop)

composition_composed:
[image]

composition_publisher and composition_subscriber:
[image]

ahcorde, 2025-05-02:
Thank you for testing

closed 2025-05-02 by ahcorde