ros2/kilted_tutorial_party #166: Executables in `examples_rclcpp_minimal_publisher`

tfoote, 2025-05-01:
## Setup
- Rmw: Cyclone
- BuildType: Binary
- Os: Windows
- Chip: Amd64

## Checks
- [ ] **Check `publisher_lambda`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run examples_rclcpp_minimal_publisher publisher_lambda
       ```
  
    
  </details>
- [ ] **Check `publisher_member_function`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run examples_rclcpp_minimal_publisher publisher_member_function
       ```
  
    
  </details>
- [ ] **Check `publisher_not_composable`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run examples_rclcpp_minimal_publisher publisher_not_composable
       ```
  
    
  </details>

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

ShahazadAbdulla, 2025-05-04:
Successfully completed all checks for Issue #166 on Windows 11 / amd64 / binary / CycloneDDS.

- Ran `ros2 run examples_rclcpp_minimal_publisher publisher_lambda`. Node started and published messages successfully.
- Ran `ros2 run examples_rclcpp_minimal_publisher publisher_member_function`. Node started and published messages successfully.
- Ran `ros2 run examples_rclcpp_minimal_publisher publisher_not_composable`. Node started and published messages successfully.

All executables ran and published data as expected.

ahcorde, 2025-05-05:
Thank you for testing

closed 2025-05-05 by ahcorde