ros2/kilted_tutorial_party #168: Executables in `examples_rclcpp_minimal_publisher`

tfoote, 2025-05-01:
## Setup
- Rmw: Fastdds
- BuildType: Binary
- Os: Windows
- Chip: Amd64

## Checks
- [ ] **Check `publisher_lambda`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run examples_rclcpp_minimal_publisher publisher_lambda
       ```
  
    
  </details>
- [ ] **Check `publisher_member_function`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run examples_rclcpp_minimal_publisher publisher_member_function
       ```
  
    
  </details>
- [ ] **Check `publisher_not_composable`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run examples_rclcpp_minimal_publisher publisher_not_composable
       ```
  
    
  </details>

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

pradyum, 2025-05-02:
**The executables are working fine**

Check `publisher_lambda`

```
C:\pixi_ws>ros2 run examples_rclcpp_minimal_publisher publisher_lambda
[INFO] [1746228867.820814900] [minimal_publisher]: Publishing: 'Hello, world! 0'
[INFO] [1746228868.320913200] [minimal_publisher]: Publishing: 'Hello, world! 1'
[INFO] [1746228868.821453700] [minimal_publisher]: Publishing: 'Hello, world! 2'
[INFO] [1746228869.321009200] [minimal_publisher]: Publishing: 'Hello, world! 3'
[INFO] [1746228869.836576900] [minimal_publisher]: Publishing: 'Hello, world! 4'
[INFO] [1746228870.320834700] [minimal_publisher]: Publishing: 'Hello, world! 5'
```

Check `publisher_member_function`

```
C:\pixi_ws>ros2 run examples_rclcpp_minimal_publisher publisher_member_function
[INFO] [1746228893.684285300] [minimal_publisher]: Publishing: 'Hello, world! 0'
[INFO] [1746228894.184470600] [minimal_publisher]: Publishing: 'Hello, world! 1'
[INFO] [1746228894.684355700] [minimal_publisher]: Publishing: 'Hello, world! 2'
[INFO] [1746228895.183709200] [minimal_publisher]: Publishing: 'Hello, world! 3'
[INFO] [1746228895.683712600] [minimal_publisher]: Publishing: 'Hello, world! 4'
[INFO] [1746228896.184291700] [minimal_publisher]: Publishing: 'Hello, world! 5'
[INFO] [1746228897.849844600] [minimal_publisher]: Publishing: 'Hello, world! 6'
[INFO] [1746228898.184358100] [minimal_publisher]: Publishing: 'Hello, world! 7'
[INFO] [1746228898.684180400] [minimal_publisher]: Publishing: 'Hello, world! 8'
[INFO] [1746228899.184020500] [minimal_publisher]: Publishing: 'Hello, world! 9'
```

Check `publisher_not_composable`

```
C:\pixi_ws>ros2 run examples_rclcpp_minimal_publisher publisher_not_composable
[INFO] [1746228921.426475600] [minimal_publisher]: Publishing: 'Hello, world! 0'
[INFO] [1746228921.936732500] [minimal_publisher]: Publishing: 'Hello, world! 1'
[INFO] [1746228922.436365800] [minimal_publisher]: Publishing: 'Hello, world! 2'
[INFO] [1746228922.935740100] [minimal_publisher]: Publishing: 'Hello, world! 3'
[INFO] [1746228923.435911000] [minimal_publisher]: Publishing: 'Hello, world! 4'
[INFO] [1746228923.936275300] [minimal_publisher]: Publishing: 'Hello, world! 5'
[INFO] [1746228924.452478000] [minimal_publisher]: Publishing: 'Hello, world! 6'
[INFO] [1746228924.936409800] [minimal_publisher]: Publishing: 'Hello, world! 7'
[INFO] [1746228925.436583100] [minimal_publisher]: Publishing: 'Hello, world! 8'
```

ahcorde, 2025-05-05:
Thank you for testing

closed 2025-05-05 by ahcorde