ros2/kilted_tutorial_party #169: Executables in `examples_rclcpp_minimal_publisher`

tfoote, 2025-05-01:
## Setup
- Rmw: Fastdds
- BuildType: Debian
- Os: Noble
- Chip: Amd64

## Checks
- [ ] **Check `publisher_lambda`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run examples_rclcpp_minimal_publisher publisher_lambda
       ```
  
    
  </details>
- [ ] **Check `publisher_member_function`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run examples_rclcpp_minimal_publisher publisher_member_function
       ```
  
    
  </details>
- [ ] **Check `publisher_not_composable`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run examples_rclcpp_minimal_publisher publisher_not_composable
       ```
  
    
  </details>

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

4-legends, 2025-05-02:
Testing this ticket

4-legends, 2025-05-02:
- Setup:
`
legends@4legends-ASUS:~$ printenv | grep RMW
RMW_IMPLEMENTATION=rmw_fastrtps_cpp
legends@4legends-ASUS:~$ printenv | grep ROS
ROS_VERSION=2
ROS_PYTHON_VERSION=3
ROS_DOMAIN_ID=0
ROS_AUTOMATIC_DISCOVERY_RANGE=SUBNET
ROS_DISTRO=kilted
`

- Check publisher_lambda
`
legends@4legends-ASUS:~$ ros2 run examples_rclcpp_minimal_publisher publisher_lambda
[INFO] [1746163931.169515283] [minimal_publisher]: Publishing: 'Hello, world! 0'
[INFO] [1746163931.669499017] [minimal_publisher]: Publishing: 'Hello, world! 1'
[INFO] [1746163932.169605658] [minimal_publisher]: Publishing: 'Hello, world! 2'
[INFO] [1746163932.669515829] [minimal_publisher]: Publishing: 'Hello, world! 3'
[INFO] [1746163933.169526058] [minimal_publisher]: Publishing: 'Hello, world! 4'
[INFO] [1746163933.669508905] [minimal_publisher]: Publishing: 'Hello, world! 5'
[INFO] [1746163934.169519255] [minimal_publisher]: Publishing: 'Hello, world! 6'
[INFO] [1746163934.669516668] [minimal_publisher]: Publishing: 'Hello, world! 7'
[INFO] [1746163935.169504546] [minimal_publisher]: Publishing: 'Hello, world! 8'
[INFO] [1746163935.669519342] [minimal_publisher]: Publishing: 'Hello, world! 9'
[INFO] [1746163936.169607142] [minimal_publisher]: Publishing: 'Hello, world! 10'
[INFO] [1746163936.669506093] [minimal_publisher]: Publishing: 'Hello, world! 11'
[INFO] [1746163937.169512422] [minimal_publisher]: Publishing: 'Hello, world! 12'
[INFO] [1746163937.669603077] [minimal_publisher]: Publishing: 'Hello, world! 13'
[INFO] [1746163938.169511209] [minimal_publisher]: Publishing: 'Hello, world! 14'
^C[INFO] [1746163938.493994141] [rclcpp]: signal_handler(signum=2)
[ros2run]: Received signal:  Interrupt
[INFO] [1746163938.494213130] [rclcpp]: signal_handler(signum=2)
`

- Check publisher_member_function
`
legends@4legends-ASUS:~$ ros2 run examples_rclcpp_minimal_publisher publisher_member_function
[INFO] [1746163858.820263361] [minimal_publisher]: Publishing: 'Hello, world! 0'
[INFO] [1746163859.320258718] [minimal_publisher]: Publishing: 'Hello, world! 1'
[INFO] [1746163859.820288747] [minimal_publisher]: Publishing: 'Hello, world! 2'
[INFO] [1746163860.320258374] [minimal_publisher]: Publishing: 'Hello, world! 3'
[INFO] [1746163860.820401533] [minimal_publisher]: Publishing: 'Hello, world! 4'
[INFO] [1746163861.320253661] [minimal_publisher]: Publishing: 'Hello, world! 5'
[INFO] [1746163861.820454787] [minimal_publisher]: Publishing: 'Hello, world! 6'
[INFO] [1746163862.320409033] [minimal_publisher]: Publishing: 'Hello, world! 7'
[INFO] [1746163862.820260313] [minimal_publisher]: Publishing: 'Hello, world! 8'
^C[ros2run]: Received signal:  Interrupt
[INFO] [1746163863.091827430] [rclcpp]: signal_handler(signum=2)
[INFO] [1746163863.091894291] [rclcpp]: signal_handler(signum=2)
`

- Check publisher_not_composable

`
legends@4legends-ASUS:~$ ros2 run examples_rclcpp_minimal_publisher publisher_not_composable
[INFO] [1746163683.189231830] [minimal_publisher]: Publishing: 'Hello, world! 0'
[INFO] [1746163683.689364464] [minimal_publisher]: Publishing: 'Hello, world! 1'
[INFO] [1746163684.189430728] [minimal_publisher]: Publishing: 'Hello, world! 2'
[INFO] [1746163684.689358866] [minimal_publisher]: Publishing: 'Hello, world! 3'
[INFO] [1746163685.189477202] [minimal_publisher]: Publishing: 'Hello, world! 4'
[INFO] [1746163685.689424103] [minimal_publisher]: Publishing: 'Hello, world! 5'
[INFO] [1746163686.189355707] [minimal_publisher]: Publishing: 'Hello, world! 6'
[INFO] [1746163686.689430955] [minimal_publisher]: Publishing: 'Hello, world! 7'
[INFO] [1746163687.189338138] [minimal_publisher]: Publishing: 'Hello, world! 8'
^C[INFO] [1746163687.376641471] [rclcpp]: signal_handler(signum=2)
[ros2run]: Received signal:  Interrupt
[INFO] [1746163687.376753895] [rclcpp]: signal_handler(signum=2)
`

[image]

closed 2025-05-02 by ahcorde