ros2/kilted_tutorial_party #17: Check action Cli

tfoote, 2025-05-01:
## Setup
- Rmw: Zenoh
- BuildType: Debian
- Os: Noble
- Chip: Amd64

## Checks
- [ ] **Check info**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run examples_rclcpp_minimal_action_server action_server_member_functions
       ```
    1.
       ```bash
       # User input in terminal 2
       ros2 action info /fibonacci
       ```
  
    **Expect**
    
    1.
       ```bash
       # stdout in terminal 2
       Action: /fibonacci
       Action clients: 0
       Action servers: 1
         /minimal_action_server
       ```
  </details>
- [ ] **Check list**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run examples_rclcpp_minimal_action_server action_server_member_functions
       ```
    1.
       ```bash
       # User input in terminal 2
       ros2 action list
       ```
  
    **Expect**
    
    1.
       ```bash
       # stdout in terminal 2
       /fibonacci
       ```
  </details>
- [ ] **Check send_goal**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run examples_rclcpp_minimal_action_server action_server_member_functions
       ```
    1.
       ```bash
       # User input in terminal 2
       ros2 action send_goal /fibonacci example_interfaces/action/Fibonacci order:\ 2
       ```
  
    **Expect**
    
    1.
       ```bash
       # stdout in terminal 2
       Waiting for an action server to become available...
       Sending goal:
           order: 2
       
       Goal accepted with ID: 0110dbd59e804374bff8e4900c3201e8
       
       Result:
           sequence:
       - 0
       - 1
       - 1
       
       Goal finished with status: SUCCEEDED
       ```
  </details>

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

khaledgabr77, 2025-05-02:
I'm testing this

khaledgabr77, 2025-05-02:
I have completed the following tests and confirmed everything works as expected.

## Checks

* [x]  **Check info**   

[image]

* [x]  **Check list**   

[image]

* [x]  **Check send_goal**   

[image]

ahcorde, 2025-05-02:
Thank you for testing

closed 2025-05-02 by ahcorde