ros2/kilted_tutorial_party #171: Executables in `examples_rclcpp_minimal_service`

tfoote, 2025-05-01:
## Setup
- Rmw: Cyclone
- BuildType: Binary
- Os: Windows
- Chip: Amd64

## Checks
- [ ] **Check `service_main`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run examples_rclcpp_minimal_service service_main
       ```
  
    
  </details>

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

mjcarroll, 2025-05-09:
Thanks for checking. One thing that could be disrupting discovery is still having a daemon running with a different middleware.  This can lead to crashes, but also weird behaviors.

If you see that again, try running with `ros2 service list --no-daemon` or doing a `ros2 daemon kill` to forcibly stop it.

closed 2025-05-09 by mjcarroll

ShahazadAbdulla, 2025-05-09:
**Update based on @mjcarroll's feedback:**
After ensuring any existing daemon was stopped (`ros2 daemon stop`), and then running `ros2 service list` (or `ros2 service list --no-daemon`), the `/add_two_ints` service **is now correctly listed** when using CycloneDDS. The direct `ros2 service call` also continued to work. This confirms the initial discovery issue was likely due to an older daemon running with a different RMW. Thanks for the tip!