ros2/kilted_tutorial_party #173: Executables in `examples_rclcpp_minimal_service`

tfoote, 2025-05-01:
## Setup
- Rmw: Fastdds
- BuildType: Binary
- Os: Windows
- Chip: Amd64

## Checks
- [ ] **Check `service_main`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run examples_rclcpp_minimal_service service_main
       ```
  
    
  </details>

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

pradyum, 2025-05-02:
**The executable is working fine**

```
C:\pixi_ws>ros2 run examples_rclcpp_minimal_service service_main
[INFO] [1746228747.516559400] [minimal_service]: request: 41 + 1
```

ahcorde, 2025-05-05:
Thank you for testing

closed 2025-05-05 by ahcorde