ros2/kilted_tutorial_party #174: Executables in `examples_rclcpp_minimal_service`

tfoote, 2025-05-01:
## Setup
- Rmw: Fastdds
- BuildType: Debian
- Os: Noble
- Chip: Amd64

## Checks
- [ ] **Check `service_main`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run examples_rclcpp_minimal_service service_main
       ```
  
    
  </details>

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

ivrolan, 2025-05-02:
Works as expected using the Ubuntu Docker Container (from Ubuntu 22.04 host amd64). I ran the client from #159 besides to test if the request are sent properly

[image]

ahcorde, 2025-05-02:
Thank you for testing

closed 2025-05-02 by ahcorde